Lora-net/LoRaMac-node #857: MLME_DEVICE_TIME returns LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT

omogenot, 2020-03-12:
Hi,

I'm implementing Release 4.4.3 and I'm currently testing the MLME_DEVICE_TIME function and see some abnormal situation (to my understanding).
Although the server did send the Server time and this is processed in the LoRaMAC through the ProcessMacCommands() function that executes the code for the SRV_MAC_DEVICE_TIME_ANS case. It even calls the SysTimeSet() function (which by the way should probably be abstracted from the LoRaMAC not to make any assumption on the way the system time is handled in the device), the LoRaMAC loop always calls the MlmeConfirm() callback with a McpsConfirm_t which MlmeRequest member contains MLME_DEVICE_TIME and Status member always contains LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT.
Is this normal and what is expected? I would have expected a LORAMAC_EVENT_INFO_STATUS_OK value as the data has been processed by the LoRaMAC without any problem.

mluis1, 2020-04-22:
Could you please check if the below commits solve the issue?
commit 0dfc7521add8244b1e91755fdcbce6c51b727f77
commit 89a4fffba057b50e1f24d3c8cbde6ac4f06bab74

mluis1, 2020-06-05:
No feedback we close the issue.

omogenot, 2020-06-05:
Actually, the test is under way and your suggestion did not help. The issue is still on.
I am trying to integrate the latest release version to check if this solves the issue, but this porting is a lengthy process....

mluis1, 2020-06-05:
As we can't reproduce the issue that you are observing it will be quite hard to help you.

Maybe, the original post doesn't exactly describe the observed issue.

Could you please try to run one of the provided examples without modifications and let us know if you also observe the issue?

The MLME_DEVICE_TIME MAC command works as expected on our different trials. It is used for the fuota-test-01, and Class B examples (classB, periodic-uplink-lpp).
We tested the correct operation of this MAC command using different network servers.

omogenot, 2020-10-28:
Hi MLuis1,

I integrated the 4.4.5 LoRaMac into the system and this solves the MLME_DEVICE_TIME timeout issue.
I therefore close the topic.

Regards,

closed 2020-10-28 by omogenot

juddvinet, 2022-05-10:
`###### ===================================== ######

Application name   : periodic-uplink-lpp
Application version: 1.3.0
GitHub base version: 4.6.0

###### ===================================== ######

DevEui      : C6-D8-BF-02-00-00-00-07
JoinEui     : 70-B3-D5-7E-D0-01-D6-3D
Pin         : 00-00-00-00

###### =========== MLME-Request ============ ######
######               MLME_JOIN               ######
###### ===================================== ######
STATUS      : OK

###### =========== MLME-Confirm ============ ######
STATUS      : Rx 2 timeout

###### =========== MLME-Request ============ ######
######               MLME_JOIN               ######
###### ===================================== ######
STATUS      : OK`

Obtain DevAddr from network for the first time LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT occurs. Where to solve the problem? Version 4.6.0 is used.