Lora-net/LoRaMac-node #862: Device stack does not support joining on LoRaWAN 1.1 Network Servers

rvolosatovs, 2020-04-02:
The current implementation of the device stack uses incorrect key pair when joining on LoRaWAN 1.1-compliant Network Servers *and* is insecure in general case.

Please see page 55 of LoRaWAN 1.1 spec.
[image]

Instead of insecure and incomplete "bruteforce" approach to determination of the key pair to use in current implementation as seen at here: https://github.com/Lora-net/LoRaMac-node/blob/0968361629561f7abc523a9a6bd79a3960983cc2/src/mac/LoRaMacCrypto.c#L1231-L1256, the device stack must record the request, which triggered the join-accept and then use either the `NwkKey` or `JSEncKey` to decrypt the join-accept received. Once the frame is decrypted, the device stack must parse the `OptNeg` field to determine further the key to use for MIC computation (either `NwkKey` or `JSIntKey`).

Currently, the device stack only works(albeit insecurely) for joining on 1.0 Network Servers.

rvolosatovs, 2020-04-03:
See also https://github.com/TheThingsNetwork/lorawan-stack/issues/2171#issuecomment-608548675

mluis1, 2020-04-14:
Thanks for the report.

This issue has been introduced when the `SecureElementProcessJoinAccept` API has been added.

We have implemented a fix which will be pushed pretty soon.
We have tested the fix using ChirpStack network server and everything looks to be functional again.

closed 2020-04-20 by mluis1