Lora-net/LoRaMac-node #864: RadioRxTimeout for unconfirmed messages

darkfader, 2020-04-09:
This happens sometimes and I don't know why.
MacCtx.MacFlags.Bits.McpsReq is set and it keeps retransmitting endlessly. Since it's an unconfirmed message, it won't get a response.
It would help as workaround if it gives up after a while.
I'm using latest develop branch.

Stack-trace:
```
#0  0x00007150 in HandleRadioRxErrorTimeout (
    rx1EventInfoStatus=rx1EventInfoStatus@entry=LORAMAC_EVENT_INFO_STATUS_RX1_TIMEOUT,
    rx2EventInfoStatus=rx2EventInfoStatus@entry=LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT)
    at ../semtech-loramac/src/mac/LoRaMac.c:1408
#1  0x0000851c in ProcessRadioRxTimeout ()
    at ../semtech-loramac/src/mac/LoRaMac.c:1450
#2  LoRaMacHandleIrqEvents () at ../semtech-loramac/src/mac/LoRaMac.c:1450
#3  LoRaMacProcess () at ../semtech-loramac/src/mac/LoRaMac.c:1662
#4  0x0000fcc6 in LmHandlerProcess ()
    at common/semtech-loramac/src/apps/LoRaMac/common/LmHandler/LmHandler.c:381
#5  0x00005a50 in task_run () at common/os/task.c:83
#6  0x00004060 in system_run () at common/lib/lib.c:82
#7  0x00002426 in main () at ./main.c:183
```

HikoSV, 2020-04-13:
I confirm the problem.
It happens to me constantly after the rejoin:  it happens at / after the clock synchronization.
I'm using develop branch (commit on Mar 10  2020)

djaeckle, 2021-05-27:
Hi all, 

thanks for the report. @darkfader , could you tell me which versions you are using?

@HikoSV could you provide some more information to reproduce this issue? Especially, which rejoin do you perform and when does the clock synchronization happen?

djaeckle, 2021-11-03:
Any feedback to the issue?

mluis1, 2021-11-18:
No feedback provided. We decide to close the issue.

In case this issue is still existing please reopen it.

closed 2021-11-18 by mluis1