Lora-net/LoRaMac-node #866: Join Request not received by the gateway

khariya, 2020-04-10:
Hello,

Here’s my system information:

- Class A device on a custom board
- Application built on feature/5.0.0 branch
- Laird Sentrius RG1xx Gateway

After [this commit](https://github.com/Lora-net/LoRaMac-node/commit/3d71aa37adb44538e804df89954e75418ff26d47) on feature/5.0.0 branch, the gateway doesn't receive the Join Request.

I can see the prints on console and the increase in current when the Join Request is sent from device.

Prior to this commit, the first Join Request was not being received by the gateway and RX 2 timeout would occur but the second Join Request was being received.

I'm unable to understand the why this occurs and what would fix this issue. Any thoughts?

EduardoPfeifer, 2020-04-10:
Hi!
You need to check this:
**CHANNEL MASK: 00FF 0000 0000 0000 0001**
The Channel Mask need to be the same as the Laird Sentrius RG1xx Gateway Radio frequencies Config .

khariya, 2020-04-15:
Thanks @EduardoPfeifer but it doesn't seem to work even after updating Channel Mask. It could be one of the things I was missing.
Any thoughts on what I should debug?

mluis1, 2020-04-20:
Which LoRaWAN region are you working with?

Please, note that one should keep the code unmodified in order to be LoRaWAN compliant.
These regions are meant to work with 64 channels. Once the end-device is connected the network server should apply the correct channel mask.
In this situation the end-device may try several times to join the network. Each join trial will send a JoinReq in every block of 8 + 1 channels.
Please refer to the regional parameters specification to see the full algorithm for JoinReq channel selection.

[image]

In case you still would like to restrict the end-device to a block of 8 channels for AU915 or US915 regions you should perform the below changes when using an 8+1 channels gateway.

```c
/**
 * Main application entry point.
 */
int main( void )
{
    ...
                mibReq.Type = MIB_PUBLIC_NETWORK;
                mibReq.Param.EnablePublicNetwork = LORAWAN_PUBLIC_NETWORK;
                LoRaMacMibSetRequestConfirm( &mibReq );

                mibReq.Type = MIB_ADR;
                mibReq.Param.AdrEnable = LORAWAN_ADR_ON;
                LoRaMacMibSetRequestConfirm( &mibReq );

#if defined( REGION_EU868 ) || defined( REGION_RU864 ) || defined( REGION_CN779 ) || defined( REGION_EU433 )
                LoRaMacTestSetDutyCycleOn( LORAWAN_DUTYCYCLE_ON );
#endif
                mibReq.Type = MIB_SYSTEM_MAX_RX_ERROR;
                mibReq.Param.SystemMaxRxError = 20;
                LoRaMacMibSetRequestConfirm( &mibReq );
// BEGIN - Workaround when using AU915/US915 with an 8 channels gateway setup
#if defined( REGION_AU915 ) || defined( REGION_US915 )
                // Enabling 1st block of 8 channels (0-7) + channel 64
                uint16_t channelMask[] = { 0x00FF, 0x0000, 0x0000, 0x0000, 0x0001, 0x0000};
                mibReq.Type = MIB_CHANNELS_MASK;
                mibReq.Param.ChannelsMask = channelMask;
                LoRaMacMibSetRequestConfirm( &mibReq );
                mibReq.Type = MIB_CHANNELS_DEFAULT_MASK;
                mibReq.Param.ChannelsDefaultMask = channelMask;
                LoRaMacMibSetRequestConfirm( &mibReq );
#endif
// END - Workaround when using AU915/US915 with an 8 channels gateway setup
                LoRaMacStart( );

                mibReq.Type = MIB_NETWORK_ACTIVATION;
                status = LoRaMacMibGetRequestConfirm( &mibReq );
    ...
}
```

khariya, 2020-04-29:
Region US915

Thank you @EduardoPfeifer and @mluis1. I was missing the correct channel mask.

My gateway has channels 8-15 enabled.
However, the node utilizes channels 0-7 only to send JoinReq even though the channel mask is set to enable all channels i.e.: FFFF FFFF FFFF FFFF 00FF 0000
I'm checking the current channel and the channel mask from `SendFrameOnChannel()`.

Why does this happen?

Restricting channels by setting the channel mask in `main()` before `LoRaMacStart()` has no effect becuase
From `main()` -> `JoinNetwork()` -> `LoRaMacMlmeRequest()` -> `ResetMacParameters()` -> `RegionInitDefaults()`  will set it to enable all channels.

If I enabled channels 0-7 on the gateway, then the node is able to send JoinReq to the gateway and  OTAA succeeds.

mluis1, 2020-04-30:
Thanks for the feedback.

Sorry, I didn't remembered that we have modified the way the channels are initialized. These modifications were necessary in order to be compliant with the LoRaWAN specifications.

But, again please note, that for fixed channel plans (AU915, CN470 and US915) restricting the channels is not recommended as the end-device will not be LoRaWAN compliant.

When using the default and recommended settings what will happen is that the Join procedure may take longer to get completed (several trials).

As your gateway is using second block of 8+1 channels it means that the first trial will always fail but, the second one should succeed.  
In case the second one doesn't succeed it will take either 6 or 8+1 other unsuccessful trials up until the stack ends up selecting either a 500 kHz (DR4) channel or a channel inside the second block (see proposed algorithm in the specification).

Please note that once joined the Network Server should automatically instruct the end-device to use the right block of channels by either providing a CFList on the JoinAccept message or by issuing LinkAdrReq MAC commands after the Join is completed.
This is the reason why it is recommended to not restrict the number of channels.

The Join procedure shouldn't happen many times in the course of the end-device operation life.  
As in this case a non standard mode of operation is being used I believe that one can accept the drawbacks that comes with it. So, one should accept that it may take longer to get an end-device  fully connected to the Network Server.

If you don't mind I close this issue.

khariya, 2020-04-30:
I think I didn't explain my issue clearly.

**Issue:**
The end-device uses channels 0-7 only, even when the channel mask is set to enable all channels.
If my gateway is using second block of 8+1 channels(or any other block except the first one), the JoinReq will never reach the gateway.

@mluis1 Can you please reopen it?

Thanks for the detailed explanation. I understand the algorithm of how channel are picked and restricting channels will make the end-device non-compliant. I'm debugging the issue, once it has been fixed, no channel will be restricted.

khariya, 2020-05-01:
This could be a possible cause for above issue:

The `ComputeNext125kHzJoinChannel()` calculates the channel based on `NvmCtx.JoinChannelGroupsCurrentIndex` and increments it (module 7) for next channel calculation.

However, from `main()` -> `JoinNetwork()` -> `LoRaMacMlmeRequest()` -> `ResetMacParameters()` -> `RegionInitDefaults()` -> `RegionUS915InitDefaults()` sets `NvmCtx.JoinChannelGroupsCurrentIndex = 0` for `INIT_TYPE_INIT` case.
This occurs each time before `ComputeNext125kHzJoinChannel()` is called.

Hence, only channels 0-7 will be selected.

Please correct me if I'm wrong or missing something.

mluis1, 2020-05-04:
I do confirm your finding.

It will require some more investigation as we did the change on https://github.com/Lora-net/LoRaMac-node/commit/3d71aa37adb44538e804df89954e75418ff26d47 commit  in order to solve another issue related to the duty-cycle management.

mluis1, 2020-05-05:
Could you please confirm that the provided fix is OK.
From our internal tests the issue should be fixed.

khariya, 2020-05-08:
Thank you for the fix, it works!

closed 2020-05-08 by mluis1

mluis1, 2020-05-11:
Thanks for the feedback.