Lora-net/LoRaMac-node #869: Portability Issues

LocatorCJSC, 2020-04-20:
We are trying to create a [MSP430FR6989 Launchpad](https://www.ti.com/tool/MSP-EXP430FR6989) -based board and there are some issues. We are in the very beginning of the process so if you are interested we will keep this post updated if new issues come along.

- We think pinName-board.h and pinName-ioe.h should be board-specific  (and the second one should be optional), i.e. move them from src/boards into src/boards/board-name (pinName-board.h MCU_PINS definition's comment even says so - STM32 Pin Names). It would be nice to have board-specific macro for defining a pin, e.g GPIO(port, pin), in the case of MSP430 it can be something like this:

```C
#define GPIO( port, pin )    ( ( uint8_t ) ( ( uint8_t ) ( ( port - 1U ) & 0x0fU ) << 4U | ( uint8_t )( pin & 0x0fU ) ) )

#define MCU_PINS \
    P1_0 = GPIO(1, 0), \
    P1_1 = GPIO(1, 1), \
    P1_2 = GPIO(1, 2), \
    P1_3 = GPIO(1, 3), \
    P1_4 = GPIO(1, 4), \
//... \
//... \
    P11_13 = GPIO(11, 13), \
    P11_14 = GPIO(11, 14), \
    P11_15 = GPIO(11, 15)
```

- This definition of PinNames type in src/system/gpio.h

```C
typedef enum
{
    MCU_PINS,
    IOE_PINS,

    // Not connected
    NC = (int)0xFFFFFFFF
}PinNames;
```

We think NC should be an unsigned type, this being -1 makes the whole enum a signed integer type, but there is a good possibility that the pin numbers are going to be bit-manipulated (moved) which is undefined behaviour for signed types.

- MSP430FR6989 does not support explicit PIN_OPEN_DRAIN outputs (it supports it automatically for example for I2C pins when the pin is configured as corresponding alternate function for I2C)

- MSP430FR6989 does not support IRQ_RISING_FALLING_EDGE interrupts and ARM-style interrupt priorities

- MSP430FR6989 supports three alternate functions for I/O, it would be nice to add some more to the existing PIN_ALTERNATE_FCT (e.g. PIN_ALTERNATE_FCT2, PIN_ALTERNATE_FCT3)

- When setting an I/O as an alternate function MSP430FR6989 must also know whether this is input or output. E.g. something like PIN_ALTERNATE_FCT | PIN_OUTPUT or PIN_ALTERNATE_FCT | PIN_INPUT which is currently impossible to do.

If you have any ideas of best approach to the mentioned issues please share.

Thanks.

LocatorCJSC, 2020-05-07:
**Update 2**

The port to MSP430FR6989 is nearly finished and it is working. The changes to other parts of the repository (other than ) are minimal.

Is Semtech interested to have this board's port in the official repository?

mluis1, 2020-06-05:
Thanks for the proposition.

Unfortunately, due to the amount of work required to maintain other platforms we cannot integrate the proposition on the official repository.

What we maybe can do is to add a link to your fork on this project Wiki pages https://github.com/Lora-net/LoRaMac-node/wiki 
We could create a page named "Other platforms support" and list the link to your fork.

What do you think?

LocatorCJSC, 2020-06-09:
In that case we will try to evaluate whether we can keep an up-to-date fork in a public repository then we will propose the fork in another issue in a later time when (or if) we are ready.

Meanwhile, in this issue, we can provide a couple of patches/pull requests that are not strictly related to MSP430 but to general portability. Mainly there were several issues because the code assumes that `int` is 32-bit. So for example this code can't be right if `int` is 16-bit or 8-bit:

```C
McSessionData[id].McGroupData.McAddr += ( mcpsIndication->Buffer[cmdIndex++] << 24 ) & 0xFF000000;
```

It should be casted:

```C
McSessionData[id].McGroupData.McAddr += ( ( uint32_t )mcpsIndication->Buffer[cmdIndex++] << 24 ) & 0xFF000000;
```

Thank you.

mluis1, 2020-06-09:
Thanks for the feedback.

Yes, it would be nice if you could provide patches/pull-request to improve this project portability.

I guess that we can then close this issue as other issues/pull-requests will be created when necessary.

closed 2020-06-09 by mluis1

techandrock, 2023-05-09:
@LocatorCJSC would you be willing to create a separate repository for the MSP430 port?