Lora-net/LoRaMac-node #871: Possible issue with dangerous pointer types convertions

LocatorCJSC, 2020-04-22:
`RtcBkupRead` expects two pointers to `uint32_t`, the function will write two 32 bit values there.

In `systime.c` there are several calls to `RtcBkupRead` where the second parameter is really pointing to `uint16_t` but is "converted" to `uint32_t *`:

`RtcBkupRead( &deltaTime.Seconds, ( uint32_t* )&deltaTime.SubSeconds );`

`deltaTime.SubSeconds` in this line above is `uint16_t` type.

This is dangerous, RtcBkupRead will treat deltaTime.SubSeconds as uint32_t and will write two additional bytes there which can corrupt things.

mluis1, 2020-06-08:
Are you Ok with patch proposed below?

```patch
diff --git a/src/system/systime.c b/src/system/systime.c
index 4fea21ba..0c4fc03a 100644
--- a/src/system/systime.c
+++ b/src/system/systime.c
@@ -119,11 +119,14 @@ SysTime_t SysTimeGet( void )
 {
     SysTime_t calendarTime = { .Seconds = 0, .SubSeconds = 0 };
     SysTime_t sysTime = { .Seconds = 0, .SubSeconds = 0 };
-    SysTime_t deltaTime;
+    uint32_t seconds;
+    uint32_t subSeconds;
 
     calendarTime.Seconds = RtcGetCalendarTime( ( uint16_t* )&calendarTime.SubSeconds );
 
-    RtcBkupRead( &deltaTime.Seconds, ( uint32_t* )&deltaTime.SubSeconds );
+    RtcBkupRead( &seconds, &subSeconds );
+
+    SysTime_t deltaTime = { .Seconds = seconds, .SubSeconds = ( int16_t )subSeconds };
 
     sysTime = SysTimeAdd( deltaTime, calendarTime );
 
@@ -141,18 +144,27 @@ SysTime_t SysTimeGetMcuTime( void )
 
 uint32_t SysTimeToMs( SysTime_t sysTime )
 {
-    SysTime_t deltaTime;
-    RtcBkupRead( &deltaTime.Seconds, ( uint32_t* )&deltaTime.SubSeconds );
+    uint32_t seconds;
+    uint32_t subSeconds;
+
+    RtcBkupRead( &seconds, &subSeconds );
+
+    SysTime_t deltaTime = { .Seconds = seconds, .SubSeconds = ( int16_t )subSeconds };
+
     SysTime_t calendarTime = SysTimeSub( sysTime, deltaTime );
+
     return calendarTime.Seconds * 1000 + calendarTime.SubSeconds;
 }
 
 SysTime_t SysTimeFromMs( uint32_t timeMs )
 {
     uint32_t seconds = timeMs / 1000;
-    SysTime_t sysTime = { .Seconds = seconds, .SubSeconds = ( int16_t )( timeMs - seconds * 1000 ) };
-    SysTime_t deltaTime = { 0 };
-    RtcBkupRead( &deltaTime.Seconds, ( uint32_t* )&deltaTime.SubSeconds );
+    uint32_t subSeconds = timeMs - seconds * 1000;
+    SysTime_t sysTime = { .Seconds = seconds, .SubSeconds = ( int16_t )subSeconds };
+
+    RtcBkupRead( &seconds, &subSeconds );
+
+    SysTime_t deltaTime = { .Seconds = seconds, .SubSeconds = ( int16_t )subSeconds };
 
     return SysTimeAdd( sysTime, deltaTime );
 }
```

LocatorCJSC, 2020-06-09:
Yes, can't test right now, but it looks okay.

mluis1, 2020-06-09:
Thanks for the feedback.

closed 2020-06-09 by mluis1