Lora-net/LoRaMac-node #873: Possible error in RegionCommonUpdateBandTimeOff function

branek, 2020-04-23:
After latest updates, callback for Duty cycle always returns 0 seconds for NextTxIn, even if sending returns LORAMAC_STATUS_DUTYCYCLE_RESTRICTED.

It looks like the problem is in this function:
```
TimerTime_t RegionCommonUpdateBandTimeOff( bool joined, Band_t* bands,
                                           uint8_t nbBands, bool dutyCycleEnabled,
                                           bool lastTxIsJoinRequest, SysTime_t elapsedTimeSinceStartup,
                                           TimerTime_t expectedTimeOnAir )
{
    TimerTime_t maxCredits = 0;
    TimerTime_t currentTime = TimerGetCurrentTime( );
    uint16_t dutyCycle = 1;
    uint16_t maxDutyCycle = 0;

    // Update bands
    for( uint8_t i = 0; i < nbBands; i++ )
    {
        // Synchronization
        dutyCycle = UpdateTimeCredits( &bands[i], joined, dutyCycleEnabled,
                                       lastTxIsJoinRequest, elapsedTimeSinceStartup,
                                       currentTime );

        // Check if the band is ready for transmission. Its ready,
        // when the duty cycle is off, or the TimeCredits of the band
        // is higher than the expected time on air.
        if( ( bands[i].TimeCredits > expectedTimeOnAir ) ||
            ( dutyCycleEnabled == false ) )
        {
            bands[i].ReadyForTransmission = true;
        }
        else
        {
            bands[i].ReadyForTransmission = false;
        }

// ######## POSSIBLE ERROR BELOW########
        // Get the band with the maximum credits
        if( ( maxCredits < bands[i].TimeCredits ) &&
            ( bands[i].TimeCredits != bands[i].MaxTimeCredits ) )
        {
            maxCredits = bands[i].TimeCredits;
            maxDutyCycle = dutyCycle;
        }
// ######## POSSIBLE ERROR ABOVE########
    }

    // Calculate the difference to the expected time on air, if the
    // expected time on air is greater than the credits of the band with highest
    // time credit value
    if( maxCredits < expectedTimeOnAir )
    {
        maxCredits = expectedTimeOnAir - maxCredits;
    }
    else
    {
        maxCredits = 0;
    }

    // Calculate the return value which shall be
    // the waiting time for the next transmission
    return maxCredits * maxDutyCycle;
}
```
Shown code above does not take in account if band[i] is enabled or not in current configuration. So, our device is connected to network where band[3] is not used for uplink but maxCredits for that band is considered valid.

After that, bottom comparison in function is not true, so maxCredits is reset to 0, and than returned value is 0. 

Not sure about right way to fix this.

djaeckle, 2020-04-23:
Hi branek,

thanks for the report.

> Shown code above does not take in account if band[i] is enabled or not in current configuration.

Could you explain briefly how you enable/disable bands? E.g. in the sample applications, all bands are enabled, but only the three default channels on one band will be used. All other bands are "enabled" but not used. It this what you mean?

In your application, do you change the definition of a band?

EduardoPfeifer, 2020-04-23:
Hi!

I had the same behavior.
Region AU915 and all bands enable. (ChirpStack setup with first sub band enable [0,1,2,3,4,5,6,7]).

###### ===========   JOINED     ============ ######
OTAA
DevAddr     :  00F13C17
DATA RATE   : DR_6

###### =========== MCPS-Request ============ ######
######           MCPS_UNCONFIRMED            ######
###### ===================================== ######
STATUS      : No channel found

###### =========== MCPS-Request ============ ######
######           MCPS_UNCONFIRMED            ######
###### ===================================== ######
STATUS      : No channel found

djaeckle, 2020-04-23:
Hi @EduardoPfeifer ,

thanks for the participation.

> Region AU915 and all bands enable. (ChirpStack setup with first sub band enable [0,1,2,3,4,5,6,7]).

What do you mean with "all bands enable". In AU915, there is only one band. Do you mean the channel mask?

Which version of the Chirpstack are you using and where do you configure this setting. Could you please describe this in detail such that we can reproduce it.

branek, 2020-04-23:
Hi @djaeckle

> Could you explain briefly how you enable/disable bands? E.g. in the sample applications, all bands are enabled, but only the three default channels on one band will be used. All other bands are [...]

You are correct, it should be "band used", not "band enabled". 

In our case, we do not do anything special. Device sends JoinRequest to network, receive RX config with channels, for EU868 region. All bands are enabled by default, but only two bands are used, with 8 channels, three from band 1, and 5 from band 0.

Now, when calculation in `RegionCommonUpdateBandTimeOff` takes place, it goes through all bands, and it does not take in account if channels from band is used or not. 

As example, in our case, when it checks band[3], it will have TimeCredits which will allow sending, it will assign that to maxCredits, and later, it will return 0 as it thinks that there is band available.

Now, when returns 0 to this function it will set `*nextTxDelay` to 0.

```
LoRaMacStatus_t RegionCommonIdentifyChannels( RegionCommonIdentifyChannelsParam_t* identifyChannelsParam,
                                              TimerTime_t* aggregatedTimeOff, uint8_t* enabledChannels,
                                              uint8_t* nbEnabledChannels, uint8_t* nbRestrictedChannels,
                                              TimerTime_t* nextTxDelay )
{
    TimerTime_t elapsed = TimerGetElapsedTime( identifyChannelsParam->LastAggrTx );
    *nextTxDelay = identifyChannelsParam->AggrTimeOff - elapsed;
    *nbRestrictedChannels = 1;
    *nbEnabledChannels = 0;

    if( ( identifyChannelsParam->LastAggrTx == 0 ) ||
        ( identifyChannelsParam->AggrTimeOff <= elapsed ) )
    {
        // Reset Aggregated time off
        *aggregatedTimeOff = 0;

        // Update bands Time OFF
        *nextTxDelay = RegionCommonUpdateBandTimeOff( identifyChannelsParam->CountNbOfEnabledChannelsParam->Joined,
                                                      identifyChannelsParam->CountNbOfEnabledChannelsParam->Bands,
                                                      identifyChannelsParam->MaxBands,
                                                      identifyChannelsParam->DutyCycleEnabled,
                                                      identifyChannelsParam->LastTxIsJoinRequest,
                                                      identifyChannelsParam->ElapsedTime,
                                                      identifyChannelsParam->ExpectedTimeOnAir );

        RegionCommonCountNbOfEnabledChannels( identifyChannelsParam->CountNbOfEnabledChannelsParam, enabledChannels,
                                              nbEnabledChannels, nbRestrictedChannels );
    }

    if( *nbEnabledChannels > 0 )
    {
        *nextTxDelay = 0;
        return LORAMAC_STATUS_OK;
    }
    else if( *nbRestrictedChannels > 0 )
    {
        return LORAMAC_STATUS_DUTYCYCLE_RESTRICTED;
    }
    else
    {
        return LORAMAC_STATUS_NO_CHANNEL_FOUND;
    }
}
```
Following call to `RegionCommonCountNbOfEnabledChannels(...)`, as there is no available channels for sending, will return `*nbRestrictedChannels > 0` and LORAMAC_STATUS_DUTYCYCLE_RESTRICTED will be returned with `*nextTxDelay` set to 0.

EduardoPfeifer, 2020-04-23:
> Do you mean the channel mask?

Yes, the channel mask.

// Initialize channels default mask
NvmCtx.ChannelsDefaultMask[0] = 0xFFFF;
NvmCtx.ChannelsDefaultMask[1] = 0xFFFF;
NvmCtx.ChannelsDefaultMask[2] = 0xFFFF;
NvmCtx.ChannelsDefaultMask[3] = 0xFFFF;
NvmCtx.ChannelsDefaultMask[4] = 0x00FF;
NvmCtx.ChannelsDefaultMask[5] = 0x0000;

> Which version of the Chirpstack are you using and where do you configure this setting. Could you please describe this in detail such that we can reproduce it.

I'm using the Gateway OS version of ChirpStack  (v3.3.0-test.6)

The file _chirpstack-network-server.toml_ is configured like:

**[network_server.network_settings]
enabled_uplink_channels=[0, 1, 2, 3, 4, 5, 6, 7]**

App -> periodic-uplink-lp

But I think the problem was here, in the Network Server.

I'm doing some tests right now and I don't know if the problem is the stack or my NS.

Soon I will update mor details.

Thanks!

EduardoPfeifer, 2020-04-23:
@djaeckle Hi! 
The problem was in my NS. 

Thanks for your time!

djaeckle, 2020-04-28:
Hi @branek,

could you provide a way to reproduce it? In the meantime, we have a potential fix. As i have problems to reproduce the issue, it would be nice if you could test it on your side:

```c
TimerTime_t RegionCommonUpdateBandTimeOff( bool joined, Band_t* bands,
                                           uint8_t nbBands, bool dutyCycleEnabled,
                                           bool lastTxIsJoinRequest, SysTime_t elapsedTimeSinceStartup,
                                           TimerTime_t expectedTimeOnAir )
{
    TimerTime_t maxCredits = 0;
    TimerTime_t currentTime = TimerGetCurrentTime( );
    uint16_t dutyCycle = 1;
    uint16_t maxDutyCycle = 0;

    // Update bands
    for( uint8_t i = 0; i < nbBands; i++ )
    {
        // Synchronization
        dutyCycle = UpdateTimeCredits( &bands[i], joined, dutyCycleEnabled,
                                       lastTxIsJoinRequest, elapsedTimeSinceStartup,
                                       currentTime );

        // Check if the band is ready for transmission. Its ready,
        // when the duty cycle is off, or the TimeCredits of the band
        // is higher than the expected time on air.
        if( ( bands[i].TimeCredits > expectedTimeOnAir ) ||
            ( dutyCycleEnabled == false ) )
        {
            bands[i].ReadyForTransmission = true;
        }
        else
        {
            bands[i].ReadyForTransmission = false;

            // Get the band with the maximum credits
            if( maxCredits < bands[i].TimeCredits )
            {
                maxCredits = bands[i].TimeCredits;
                maxDutyCycle = dutyCycle;
            }
        }
    }

    // Calculate the difference to the expected time on air, if the
    // expected time on air is greater than the credits of the band with highest
    // time credit value
    if( maxCredits < expectedTimeOnAir )
    {
        maxCredits = expectedTimeOnAir - maxCredits;
    }
    else
    {
        maxCredits = 0;
    }

    // Calculate the return value which shall be
    // the waiting time for the next transmission
    return maxCredits * maxDutyCycle;
}
```

branek, 2020-04-28:
Hi @djaeckle, thanks for your help.

> 
> could you provide a way to reproduce it?

Of course. 
 - Please, make sure you are using EU868 band.
 - To speed up thing, in Region.h, change definition of ` #define DUTY_CYCLE_TIME_PERIOD` to be 10 times shorter, like 360000. (bug will appear with original settings but you need to wait longer).
 - Configure your device to send messages really fast (I send 1 32 byte message every 5 seconds).
 - After few successfully sent messages, device will hit duty cycle, and you will be able to observe bug.

Regarding fix, unfortunately it does not change behavior. If I could suggest dirty fix which work for me, maybe it will help you to understand issue:
```
diff --git a/Middlewares/Third_Party/LoRa/mac/region/RegionCommon.c b/Middlewares/Third_Party/LoRa/mac/region/RegionCommon.c
index 8b924db..89d75a4 100644
--- a/Middlewares/Third_Party/LoRa/mac/region/RegionCommon.c
+++ b/Middlewares/Third_Party/LoRa/mac/region/RegionCommon.c
@@ -252,26 +252,28 @@ TimerTime_t RegionCommonUpdateBandTimeOff( bool joined, Band_t* bands,
                                        lastTxIsJoinRequest, elapsedTimeSinceStartup,
                                        currentTime );
 
-        // Check if the band is ready for transmission. Its ready,
-        // when the duty cycle is off, or the TimeCredits of the band
-        // is higher than the expected time on air.
-        if( ( bands[i].TimeCredits > expectedTimeOnAir ) ||
-            ( dutyCycleEnabled == false ) )
-        {
-            bands[i].ReadyForTransmission = true;
-        }
-        else
-        {
-            bands[i].ReadyForTransmission = false;
-        }
+        // ##### Fix for duty cycle #####
+        // All unused bands are disabled. Now assign maxCredits only
+        // from used bands
 
         // Get the band with the maximum credits
         if( ( maxCredits < bands[i].TimeCredits ) &&
-            ( bands[i].TimeCredits != bands[i].MaxTimeCredits ) )
+            ( bands[i].TimeCredits != bands[i].MaxTimeCredits ) &&
+			( bands[i].ReadyForTransmission == true) )
         {
             maxCredits = bands[i].TimeCredits;
             maxDutyCycle = dutyCycle;
         }
+
+        // Check if the band is ready for transmission. Its ready,
+        // when the duty cycle is off, or the TimeCredits of the band
+        // is higher than the expected time on air.
+        if( ( bands[i].TimeCredits < expectedTimeOnAir ) &&
+            ( dutyCycleEnabled == true ) )
+        {
+            bands[i].ReadyForTransmission = false;
+        }
+        // ##### End of fix #####
     }
 
     // Calculate the difference to the expected time on air, if the
@@ -502,6 +504,32 @@ LoRaMacStatus_t RegionCommonIdentifyChannels( RegionCommonIdentifyChannelsParam_
         // Reset Aggregated time off
         *aggregatedTimeOff = 0;
 
+        // ##### Fix for duty cycle #####
+        //disable all bands for transmission
+        for( uint8_t i = 0; i < identifyChannelsParam->MaxBands; i++ )
+        {
+        	identifyChannelsParam->CountNbOfEnabledChannelsParam->Bands[i].ReadyForTransmission = false;
+        }
+
+        //Part bellow is stolen from RegionCommonCountNbOfEnabledChannels()
+        //It marks band of each enabled channel as
+        //ready for transmission.
+        for( uint8_t i = 0, k = 0; i < identifyChannelsParam->CountNbOfEnabledChannelsParam->MaxNbChannels; i += 16, k++ )
+        {
+            for( uint8_t j = 0; j < 16; j++ )
+            {
+                if( ( identifyChannelsParam->CountNbOfEnabledChannelsParam->ChannelsMask[k] & ( 1 << j ) ) != 0 )
+                {
+                	//If channel not 0, enabled?
+                    if( identifyChannelsParam->CountNbOfEnabledChannelsParam->Channels[i + j].Frequency != 0 )
+                    { // Check if the channel is enabled and mark band ready for transmission.
+                    	identifyChannelsParam->CountNbOfEnabledChannelsParam->Bands[identifyChannelsParam->CountNbOfEnabledChannelsParam->Channels[i + j].Band].ReadyForTransmission = true;
+                    }
+                }
+            }
+        }
+        // ##### End of fix #####
+
         // Update bands Time OFF
         *nextTxDelay = RegionCommonUpdateBandTimeOff( identifyChannelsParam->CountNbOfEnabledChannelsParam->Joined,
                                                       identifyChannelsParam->CountNbOfEnabledChannelsParam->Bands,

```
This works for me, but I am quite sure there is a better way for solving this. 

If you need anything else, I'm here to help.

djaeckle, 2020-05-05:
Thanks for the verification and your code. This is the latest version:

```c
TimerTime_t RegionCommonUpdateBandTimeOff( bool joined, Band_t* bands,
                                           uint8_t nbBands, bool dutyCycleEnabled,
                                           bool lastTxIsJoinRequest, SysTime_t elapsedTimeSinceStartup,
                                           TimerTime_t expectedTimeOnAir )
{
    TimerTime_t minTimeToWait = TIMERTIME_T_MAX;
    TimerTime_t currentTime = TimerGetCurrentTime( );
    uint16_t dutyCycle = 1;
    uint8_t validBands = 0;

    // Update bands
    for( uint8_t i = 0; i < nbBands; i++ )
    {
        // Synchronization
        dutyCycle = UpdateTimeCredits( &bands[i], joined, dutyCycleEnabled,
                                       lastTxIsJoinRequest, elapsedTimeSinceStartup,
                                       currentTime );

        // Check if the band is ready for transmission. Its ready,
        // when the duty cycle is off, or the TimeCredits of the band
        // is higher than the expected time on air.
        if( ( bands[i].TimeCredits > expectedTimeOnAir ) ||
            ( dutyCycleEnabled == false ) )
        {
            bands[i].ReadyForTransmission = true;
            // This band is a potential candidate for an
            // upcoming transmission, so increase the counter.
            validBands++;
        }
        else
        {
            // In this case, the band has not enough credits
            // for the next transmission.
            bands[i].ReadyForTransmission = false;

            if( bands[i].MaxTimeCredits > expectedTimeOnAir )
            {
                // The band can only be taken into account, if the maximum credits
                // of the band are higher than the expected time on air.
                // We calculate the minTimeToWait among the bands which are not
                // ready for transmission and which are potentially available
                // for a transmission in the future.
                minTimeToWait = MIN( minTimeToWait, ( expectedTimeOnAir - bands[i].TimeCredits ) * dutyCycle );
                // This band is a potential candidate for an
                // upcoming transmission (even if its time credits are not enough
                // at the moment), so increase the counter.
                validBands++;
            }
        }
    }

    if( validBands == 0 )
    {
        // There is not one valid band which can be used for the
        // transmission. Increase the value DUTY_CYCLE_TIME_PERIOD.
        return TIMERTIME_T_MAX;
    }
    // Calculate the return value which shall be
    // the waiting time for the next transmission
    return minTimeToWait;
}
```

It would be nice if you could run it again and let me know the outcome.

branek, 2020-05-06:
Hi @djaeckle,

Thanks for the effort. First, let me tell you that our application is slightly different regarding implementation compared to examples from here. We are using FreeRTOS, but nothing is changed in stack.

Right now your fix is working, but I see same erratic behavior in one corner case: when LORAMAC_STATUS_DUTYCYCLE_RESTRICTED is returned for the first time, everything is ok, time returned is ok, and we are starting timer with returned duty cycle time as expiration period.

If we call resend immediately after timer expires, we got error. But if we wait for 20 to 30 miliseconds, we got everything as expected. This behavior started after duty cycle handling refactoring. Let me do some more tests about this.

djaeckle, 2020-05-07:
HI @branek,

many thanks for your feedback. It really helps us to isolate the issue and provide fixes.

> If we call resend immediately after timer expires, we got error. 

Do you mean with "error" that you get a status of `LORAMAC_STATUS_DUTYCYCLE_RESTRICTED`?

branek, 2020-05-08:
Hi @djaeckle,

Yes, error is the same as described in first post, `LORAMAC_STATUS_DUTYCYCLE_RESTRICTED` is returned but time to wait is 0. Not able to isolate cause yet, but it looks like some kind of race condition.

To resume, steps how we reproduce issue on our side after your latest diff:
1. Device JOIN to network.
2. Device sending messages in high rate (one message of 32 bytes every 5 seconds, confirmed message)
3. After some time, we hit duty cycle limit, with `LORAMAC_STATUS_DUTYCYCLE_RESTRICTED` and valid time.
4. We start timer with expiration time equal to time received in previous step.
5. When timer expires, we call LmHandlerSend() as soon as possible to resend previously failed message.
6. Stack return LORAMAC_STATUS_DUTYCYCLE_RESTRICTED with time value of 0.

But if, in step 4, we add 1000 ms longer waiting time, everything is working ok. We are reserved about cause, maybe there is a problem on our side. We are running LmHandlerProcess() from idle task, and sending messages from higher priority task. It could be that we call LmHandlerSend() too soon, but in previous stack versions that was rock solid, without issues.

mluis1, 2020-05-18:
Hi @branek 

We have just pushed a refactored and improved implementation of the duty-cycle-management.
From our internal tests we think that the observed corner case should be fixed by these commits.

Could you please verify this new proposition and let us know if it is the case?

branek, 2020-05-18:
Hello @mluis1 

Thanks for your help. We are able to confirm that on our tests observed bug is fixed. We will continue additional testing, but we think this issue is ready to be closed and marked as solved.

mluis1, 2020-05-18:
Nice, thanks for the feedback.

closed 2020-05-18 by mluis1