Lora-net/LoRaMac-node #874: RADIO_BUSY stuck HIGH

ahiregange, 2020-05-04:
Hi,

I am using a custom board with SX1262 with a SiLabs Gecko MCU and been successful in getting join/uplink/downlink etc. working very consistently with multiple units.

Now I am working on robustness and found that always within about 4 hours of operation with transmit interval of about 6minutes, the stack was getting stuck. After some debugging I found that this was always due to getting stuck in SX126xWaitOnBusy(), presumably because the GPIO was not coming LOW from HIGH after some operation.

For now I have made this change to detect an abnormally long call of this function and break out of it. With this now the unit has not got stuck for 4+ days so it seems the workaround is good.

void SX126xWaitOnBusy( void )
{
    uint32_t busy = 1;
    int count = 0;
    while( busy == 1 )
    {
        busy = GpioRead( &SX126x.BUSY );
        count++;
        if (count > 100000)
        {
            printf ("Break out of infinite BUSY\n");
            break;
        }
    }
}

1) Has any such issue been reported on any other platform?
2) Any thoughts on why BUSY could be stuck HIGH forever? For now the workaround is okay but I am worried whether it could be hiding some other issue which is causing BUSY to be stuck.

Thanks,
Arun

matthieuantoine, 2020-06-10:
Hi Arun,

Thanks for reporting this issue.

Can you please let me know what the state of the chip is before the call to `SX126xWaitOnBusy` that fails (sleep, etc.)?

Thanks.
_Matthieu

j-student-m, 2020-07-23:
Hi,

I've also had a problem with the busy line not coming down. I found out it had something to with the sleepmode of the chip.
According to "SX1261/2 Long Range, Low Power, sub-GHz RF Transceiver " HS 9.3 the chip busy line will remain High when it is in sleep. So the chip is probably in sleep mode. I have aplied the following fix:

In the file "sx126x.c" there is the following function:

void SX126xSetStandby( RadioStandbyModes_t standbyConfig )
{
   
    if( standbyConfig == STDBY_RC )
    {
        SX126xSetOperatingMode( MODE_STDBY_RC );
    }
    else
    {
        SX126xSetOperatingMode( MODE_STDBY_XOSC );
    }
 **SX126xWriteCommand( RADIO_SET_STANDBY, ( uint8_t* )&standbyConfig, 1 );**
}

You should change it to this

void SX126xSetStandby( RadioStandbyModes_t standbyConfig )
{
    **SX126xWriteCommand( RADIO_SET_STANDBY, ( uint8_t* )&standbyConfig, 1 );**
    if( standbyConfig == STDBY_RC )
    {
        SX126xSetOperatingMode( MODE_STDBY_RC );
    }
    else
    {
        SX126xSetOperatingMode( MODE_STDBY_XOSC );
    }
}

Because when coming out of sleep mode the chip needs to be set in standby with the standby command. Because the operatingmode variable is set to standby, before it actually sends the command to the chip, the code will be stuck in SX126xCheckDeviceReady( ) when checking the busypin

cheers

mluis1, 2020-09-01:
Should have been fixed with changes done to fix issue #927.

closed 2020-09-01 by mluis1