Lora-net/LoRaMac-node #875: Default max EIRP wrong on EU868

Pirakutu, 2020-05-05:
Hi guys,

I just wanted to optimize the RF performance of my device and realized that the default EU868_DEFAULT_MAX_EIRP is wrongly defined @ 16.0 dBm, when it actually can be 14 + 2.15 -> 16.15 dBm. Because of this, the actual power the radio chip is configured to at the end is not 14 dBm but 13 dBm (floor calculation inside RegionCommon -> from 13.85 to 13)!

Cheers,

Andres

mluis1, 2020-10-21:
The `EU868_DEFAULT_MAX_EIRP` is defined as per the regional parameters specification.

> By default, the Max EIRP is considered to be +16dBm. If the end-device cannot achieve 16dBm EIRP, the Max EIRP SHOULD be communicated to the network server using an out-of-band channel during the end-device commissioning process.

Applying `floor` calculation is safer as we ensure that the 16dBm value is never exceeded with 2.15dB antenna gain.

Also depending on your PCB design, antenna matching and used antenna nothing prevents you to modify the `void <RadioName>SetRfTxPower( int8_t power )` function located under radio specific board file (i.e: `sx1261mbxbas-board.c`) to fulfill your requirements. 

However, please keep in mind to always comply with national RF regulations (i.e: FCC, ETSI, etc...).

closed 2020-10-21 by mluis1

Aboubar, 2021-03-06:
> The `EU868_DEFAULT_MAX_EIRP` is defined as per the regional parameters specification.
>
>
> Applying `floor` calculation is safer as we ensure that the 16dBm value is never exceeded with 2.15dB antenna [...]

Okay, thank you for this information.