Lora-net/LoRaMac-node #888: OTAA Activation Fails

mcu-guy, 2020-05-31:
Hi, 

I am trying to get my node to join the LoRa NW in the US915 region. I am using the default data rate as DR_5, unconfirmed message type. I am seeing my node sending the Join Req packets from the Gateway logs but the Join Response is never sent back, when I debugged further, I see that the MLME call back for Join Req fails with status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT. 
I am using the STM32's I-CUBE-LRWAN package and MachineQ gateway. 
Can somebody help me understand what could be the potential problem here?

ConnyBusy, 2020-06-01:
Note that there is no DR_5 for the uplink in the US915 Band. Uplink data rates are defined as follows:
DR_0 ==> SF10BW125
DR_1 ==> SF9BW125
DR_2 ==> SF8BW125
DR_3 ==> SF7BW125
DR_4 ==> SF8BW500

mluis1, 2020-06-05:
As @ConnyBusy mentioned DR_5 data  rate doesn't exist for the US915 region.

From your description it is not clear if the network server is really sending the JoinAccept message.
In case the network server doesn't send the JoinAccept message it means most certainly that the device identity is not correct. Please check that DevEUI, JoinEUI (AppEUI-1.0.x), NWK_KEY(APP_KEY-1.0.x) are correctly set.

The `LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT` status here means that there was no down link received on either Rx1 or Rx2 windows. Thus, the join procedure couldn't be completed.

closed 2020-06-05 by mluis1