Lora-net/LoRaMac-node #889: Which network server used for testing?

WinXprt, 2020-06-01:
Couldn't find the which network server used for the testing. 
Let me know where can I get it?

altishchenko, 2020-06-01:
What do you mean by testing? Running official tests or experimenting with your equipment? If the latter is the case, depends upon your equipment - somebody tests against TTN, some against Everynet, some people run their own servers, like this gotthardp/lorawan-server.

WinXprt, 2020-06-02:
I am looking for fuota-test1 application
How this application is being tested on SAMR34 devices like SAMR34 Xplained Pro? which network server is used for testing?

As SAMR34 Xplained Pro doesn't have external flash, does this application uses internal flash(256kb) of SAMR34?

If the application is being explained with at least one network server this will gives much clarity.

That helps a lot.

altishchenko, 2020-06-02:
Oh, I see. Well, the application provided is just a proof of concept, example. It show how one can switch between Class A normal operation into Class C multicast big data chunk download.
The real FUOTA process itself depends on multiple things and mostly on MCU and board capabilities. More often than not, there must be a small special "boot loader" that starts first and checks for valid 'images' in either internal or external flash, picking up the appropriate and continuing the boot process with it. For that, it is required that your flash be capable of holding two complete firmware images and bootloader itself. You can picture it as:
```
[Boot Loader, MCU starts here]:[Memory bank 0, Firmware 1]:[Memory bank 1, Firmware 2]
```
Depending on the MCU capabilities the code in Firmware is either ran directly from its physical location or being copied upon startup to a code running memory region. 
You can get more insights in different approaches on the [mcuboot design](https://juullabs-oss.github.io/mcuboot/design.html) page, like one of the most popular projects.

mluis1, 2020-06-05:
In order to test the FUOTA protocol defined by the LoRa-Alliance we have used Chirpstack network server.

As said by @altishchenko the provided example on this project is just to test the protocol.
The received data is stored on a RAM buffer.

In order to have full firmware upgrade one must implement a bootloader as well as the functions to write the MCU flash memory.

closed 2020-06-05 by mluis1