Lora-net/LoRaMac-node #890: Uplink Frame Counter initial value

DoubleCheeseBacon, 2020-06-02:
Hi all,

Should the Uplink frame counter start at 0 ?
Because "LoRaMacCryptoGetFCntUp" return the FCntUp + 1 ....
So first frame get FCntUp = 1 instead of 0
Am I missing something or it is a little bug ?

Thanks

altishchenko, 2020-06-02:
This is a long unfortunate story of the LoRaWAN specification. Here is a quote from 1.0.3: "After a JoinReq – JoinAccept message exchange or a reset for a personalized end-device, the frame counters on the end-device and the frame counters on the network server for that end-device are reset to 0. Subsequently FCntUp and FCntDown are incremented at the **sender side by 1 for each new** data frame sent in the respective direction." The 1.1 spec made things even more unclear. Most implementers treat this as the first frame to be sent is the first **new** frame, so the counter should be 1.
Because of this lack of clarity everybody out there handles 0s and 1s the same, i.e. first frame after reset.

DoubleCheeseBacon, 2020-06-03:
I agree with you it's not clear at all.

kjendal, 2020-06-03:
This will be clarified in 1.0.4 and 1.1.1.  The original intent (and so clarified) is for FCnt to begin at zero (the first frame is sent with FCnt=0).
That said, there is small impact (worst case is a single discarded frame).

closed 2020-06-03 by mluis1