Lora-net/LoRaMac-node #893: Projected 1.0.4 publication date

altishchenko, 2020-06-04:
@mluis1 Miguel, do you have a projected date for 1.0.4 publication? I can see Daniel (@djaeckle) is pushing lots of stuff getting ready for 1.0.4, but the spec is not on resource hub yet. You guys have access to the drafts and I am curious if there are any critical changes that will require network server side modifications (apart from 0/1 FCnt story) ?
I am a contributor to gotthardp/lorawan-server project and we use it in production in Softline, so my interest is quite practical.

Cheers,
Alex

mluis1, 2020-06-04:
I don't know when the LoRaWAN 1.0.4 specification will be publicly released. But, I would expect it to be released before the end of the year.

If I'm not mistaken I guess that all LoRa-Alliance members have access to the draft document.

As we have finalized the 1.0.4 update we decided to push the code in order to ease the network servers update implementation checks.

The changes from 1.0.3 to 1.0.4 specification are mainly:
* Integration of 1.0.3 ERRATA notes
* Back porting as much as possible the specified features from 1.1.1 (Excludes 1.1.1 new MAC commands as well as the new key derivation scheme)
* Clarifications

altishchenko, 2020-06-04:
@mluis1 Ah well, we don't have access unfortunately. Thank you anyway!

closed 2020-06-04 by altishchenko