Lora-net/LoRaMac-node #894: Join request MIC check failed

lucaspeixotot, 2020-06-04:
Hello, I'm making a porting process to use LoRaMac-node on a different platform. I already tested node in ABP mode class A and B and everything worked well. Now I'm trying to use the node in OTAA mode and I'm having some issues. The problem is that when I send a join request I received an error message in network server saying: **Join request MIC check failed for DEV_EUI:**. I think that this problem is about the keys that I need to set before the join process. I'm confused about the join process, about what data I must have stored in the device to OTAA join process. I'm using [v4.4.4](https://github.com/Lora-net/LoRaMac-node/releases/tag/4.4.4) tag version of LoRaMac-node. This version is supposed to be a version for LoRaWAN v1.0.3, but I think that the v4.4.4 version is working v1.0.3 and v1.1 as well.

First of all, in [quick-start guide](http://stackforce.github.io/LoRaMac-doc/_q_u_i_c_k__s_t_a_r_t__g_u_i_d_e.html#QUICKSTART_OTA_ACTIVATION) we need just DevEui and JoinEui data previously stored in end-device. However, if I take a look for examples code we can see that the example uses beyond the DevEui and JoinEui it uses SE_PIN(I don't know exactly what is it). This is strange because the example is for v1.0.0 and these keys don't correspond to it. If I open LoraWAN specification V1.0.3 the document is clear:
[image]

Look that the v1.0.3 doesn't even say about a JoinEui data and I realized that JoinEui is data for v1.1. As I'm thinking that the LoRaMAC-node implementation is for v1.1 and made to works in v1.0.3 I opened the LoraWAN specification V1.1 specification to see what exactly I need to set for join process. So I found that:
[image]

My network server is in EveryNet and I have the following data available:
- Device EUI
- Application EUI
- Application Key

Along with this issue description, I show a lot of different procedures to be executed in join process, but I don't know exactly what I must have stored in device to the OTAA join process. May someone clarify this for me?

EduardoPfeifer, 2020-06-04:
Hi @lucaspeixotot 
When are you setting this parameters? Some of then is impossible to change after a LmHandlerInit.

You need to check the se-identity.h 

"SOFT_SE_KEY_LIST -> APP_KEY"

altishchenko, 2020-06-04:
@lucaspeixotot Hello Lucas, you are correct in understanding that you only need DevEUI, AppEUI(a.k.a. JoinEUI) and AppKey. The PIN field you are quoting is used to access the real hardware secure element, much like SIM pin is used to unlock the phone. Unless you have LR1110 you don't need it. JoinEUI is defined in se-identity.h under the name LORAWAN_JOIN_EUI.
To test it out, try setting the APP_KEY and the **NWK_KEY** to the same value, as there is some naming collision between LoRaWAN 1.0.3 and LoRaWAN 1.1 versions of the standard.
Another possible source of problem can be that sometimes EUIs or Keys are **reversed** on the network server, I am just not sure about the Everynet. You can just try entering your key _backwards_ to check:
```
{0x01, 0x02, 0x03, ..., 0x10 } -> {0x10, ..., 0x03, 0x02, 0x01}
```

EduardoPfeifer, 2020-06-04:
Everynet use 1.0.x DEVICES

```
/*!                                                                                                     \
             * Network root key                                                                                     \
             * WARNING: FOR 1.0.x DEVICES IT IS THE \ref LORAWAN_APP_KEY                                            \
             */                                                                                                     \
            .KeyID    = NWK_KEY,                                                                                    \
            .KeyValue = { 0x4B, 0x41, 0x70, 0x70, 0x6C, 0x69, 0x63, 0x61, 0x74, 0x69, 0x6F, 0x6E, 0x20, 0x4B, 0x65, \
                          0x79 }, 
```

lucaspeixotot, 2020-06-04:
I merged information passed by you guy, @EduardoPfeifer and @altishchenko, and I just set MIB_NWK_KEY equal to MIB_APP_KEY. Thanks, everything is working now and is not needed to set these keys: MIB_F_NWK_S_INT_KEY and MIB_S_NWK_S_INT_KEY and MIB_NWK_S_ENC_KEY.

closed 2020-06-04 by lucaspeixotot