Lora-net/LoRaMac-node #895: I cube Lorawan Stack V1.3.1 user Code Not Working/Updating

iamcalledusman, 2020-06-05:
#Hi All 

Hope you all are doing well! . I have a question , i will highly appreciate for the answer.
Currently i am using I cube Lorawan Stack V1.3.1 its compiling and working well with my Nucleo Board+ Sx1262 . But when i copy pasted the working code from another project in which HAL libraries used . In this project i interfaced the DS1307/TLV493D , read the data from it and displayed on UART perfectly .

After copy pasting all the code(DS1307/TLV493D) into the I cube Lorawan with all the variables with zero error . Program failed to work . I debugged the issue . Which i found is that 

In working code of (DS1307/TLV493D)  internal registers of I2C were updating in debug mode . I have attached the screen shot of it . 

But after copy pasting program get stuck at this command 

if(HAL_I2C_IsDeviceReady(&hi2c1 ,0x00 ,1,10)== HAL_OK){
		
		HAL_GPIO_TogglePin(GPIOA,GPIO_PIN_5);
			
	}
I debugged the issue , in which i found that internal registers of I2c are not updating at all . I also tried UART code / Timer Code which was working in separate project but doing same behavior as mentioned above for the case of I2C .Screen Shot is also attached which is showing internal registers are not updating. 

Note * 
I have also configured the I2C clock as well in system configuration function . I have also attached the main.c files for both the cases .

Thanks in Advance
Best Regards,

](

)
[image]
[image]
[Icubelorawan_code_pasted.txt](https://github.com/Lora-net/LoRaMac-node/files/4735374/Icubelorawan_code_pasted.txt)
[working_Code_i2c.txt](https://github.com/Lora-net/LoRaMac-node/files/4735375/working_Code_i2c.txt)

mluis1, 2020-06-05:
en.i-cube_lrwan is a fork of this project made by ST.

The issue that you are reporting looks to be related to the fork itself.
As we don't know what are the differences between ST fork and this project we can't really help.

I would recommend that you contact ST support and ask them for help.

closed 2020-06-05 by mluis1

iamcalledusman, 2020-06-05:
> en.i-cube_lrwan is a fork of this project made by ST.
> 
> The issue that you are reporting looks to be related to the fork itself.
> As we don't know what are the differences between ST fork and this project we can't really help.
> 
> I would recommend that you contact ST support and ask them for help.

Thanks for the prompt response and for the right direction , let me ask the ST for the help :).

Thank You !

altishchenko, 2020-06-05:
@iamcalledusman FYI, your screenshots show that I2C is not initialized. I can try  helping you, if you want, but do so by e-mail (atishch@gmail.com), not in this thread.

iamcalledusman, 2020-06-05:
> @iamcalledusman FYI, your screenshots show that I2C is not initialized. I can try helping you, if you want, but do so by e-mail ([atishch@gmail.com](mailto:atishch@gmail.com)), not in this thread.

thank you !