Lora-net/LoRaMac-node #896: Lorawan Class B supported Servers

iamcalledusman, 2020-06-05:
Hi All,
I am new to lorawan world , can you please guide me which server supports Class B ?
I have tried the chirp stack but its too complicated . Do you know any server like loriot server so that i can easily test my Firmware with that .

Thanks in advance

BR

mluis1, 2020-06-05:
This question is not related to an issue with this project.

This kind of questions are better suited to be asked on a forum.
Please find below an example of forum where such question will get better support.
https://lora-developers.semtech.com/knowledge-base/forum/

Personally, I use Chirpstack in order to test the Class B implementation. Which in my opinion is pretty simple to setup and to use. If you follow the Chirpstack provided documentation it should be quite straight forward.
Use following link to get started https://www.chirpstack.io/guides/docker-compose/

closed 2020-06-05 by mluis1

iamcalledusman, 2020-06-05:
> This question is not related to an issue with this project.
>
> This kind of questions are better suited to be asked on a forum.
> Please find below an example of forum where such question will get [...]

thanks for help and response , i really appreciate it