Lora-net/LoRaMac-node #898: Warning[Pa118]: mixing boolean and non-boolean types in a comparison results in the boolean being promoted

Xeenych, 2020-06-05:
IAR Compiler produces the following warning

Warning[Pa118]: mixing boolean and non-boolean types in a comparison results in the boolean being promoted D:\Projects\LoRaArtur\amdk_F103\LoraMac\mac\LoRaMac.c 1128 

https://github.com/Lora-net/LoRaMac-node/blob/d3791e2a00598452eb34290c69447727dfb3fe52/src/mac/LoRaMac.c#L1130

The same for line #1129

EduardoPfeifer, 2020-06-05:
> ( macMsgData.FHDR.FCtrl.Bits.Ack == true ) ||

I recomend:

`( macMsgData.FHDR.FCtrl.Bits.Ack != 0 )`

mluis1, 2020-06-05:
Thanks for the finding.

Would you be willing to propose a pull-request?

@EduardoPfeifer proposition is the right way to go.

Xeenych, 2020-06-05:
I wonder if I've done ot right
https://github.com/Lora-net/LoRaMac-node/pull/899

mluis1, 2020-06-08:
Yes, it looks good. Thanks.

PR #899 (2020-06-08): Update LoRaMac.c
Closes #898
Remove IAR C++ Compiler warning
Warning[Pa118]: mixing boolean and non-boolean types in a comparison results in the boolean being promoted
closed 2020-06-08 by mluis1