Lora-net/LoRaMac-node #906: IAR C++ compiler errror:  cast to uint8_t[]

Xeenych, 2020-06-08:
IAR C++ Compiler produces the following error
Error[Pe2363]: cast to incomplete array type "uint8_t []" is not allowed D:\Projects\LoRaArtur\amdk_L151\LoraMac\apps\LoRaMac\common\LmHandler\packages\FragDecoder.c 283 

https://github.com/Lora-net/LoRaMac-node/blob/0b7dead2c0b678891cef1751872ca6a89d11890b/src/apps/LoRaMac/common/LmHandler/packages/FragDecoder.c#L283

There shoild be something like this:
```c++
    static uint8_t dummy_data[] = {0xFF};
    FragDecoder.Callbacks->FragDecoderWrite( i, dummy_data, 1 );
```
dummy_data is made static cause I don't know if FragDecoderWrite may use some sort of DMA or IRQ.

mluis1, 2020-06-08:
I guess that there is no need to make the variable static here. We are just initializing every byte to `0xFF`.

Could you try to change the the line as follows ?
Please let us know if the warning disappears.

```c
FragDecoder.Callbacks->FragDecoderWrite( i, ( uint8_t[1] ){ 0xFF }, 1 ); 
```
I don't have access to IAR C++ compiler thus, I can't verify myself.

Xeenych, 2020-06-08:
This doesn't work

mluis1, 2020-06-08:
Thanks, 
But, I don't see why not?

It works in C language. Also note that this is a C language project.

closed 2020-06-08 by mluis1