Lora-net/LoRaMac-node #907: How low can B-L072Z-LRWAN1 low power be?

rebakker, 2020-06-09:
Hi,
I've been designing a LoRa node based on the B-L072Z-LRWAN1 for some time and wanted to check the power consumption. Which I find somewhat high. So I changed back to the original application from LoRaMac-node-master (4.4.3).
I have now taken the used LoRaMac-node-master and used Keil uVision 5 to compile the Class A application for the B-L072Z-LRWAN.
I changed the _Commissioning.h_ file and changed _UsbIsConnected = false;_ in _void BoardInitMcu( void )_ in board.c. This results in a working LoRa node.

To measure the current I disconnected **SB6**, placed a external +3.3V on **CN13** and use a current meter to measure the total current for **VDD_USB_LRA**, **VDD_MCU_LRA** and **VDD_RF_LRA**. 
**JP9** is set to 1-2 so **PA12** controls the TCXO power. **JP10** is also in the 1-2 position, and JP5 is not selected.

The lowest measured current is in Stop mode about 500uA. This is much higher than the expected 2uA from the Murata datasheet (page 11).
What am I missing here?

mluis1, 2020-06-09:
First I would like to inform that this kind of questions are better suited to be asked on a forum.
Please find below an example of forum where such question will get better support.
https://lora-developers.semtech.com/knowledge-base/forum/

The 2uA value is the value to be expected for MCU+Radio only.
I don't know how you are measuring the power consumption. But, from the description I guess that you are also measuring the ST-Link, LED, ... power consumption.

The best way to compute the power consumption is to use an oscilloscope where 2 probes are connected to each side of an 100 Ohm resistor in series with Murata module power supply.
Then on the oscilloscope you need to add a mathematical subtraction function over the 2 probe signals.
Then the power consumption will be the measured voltage on the mathematical signal divided by the used resistance.

You should find something similar to the below screenshot. The displayed signal is the result of the mathematical function.
[image]

closed 2020-06-09 by mluis1

altishchenko, 2020-06-09:
If you are using Keil, you can at least make sure your board does go into stop mode proper (function LpmEnterStopMode()), /*Off mode is not implemented for this board as I can see from the code*/. Put a breakpoint at the entry to this function (found in board.c file) or watch in 'View' -> System Viewer -> Core Peripherals -> 'System Control and Configuration' the SLEEPDEEP flag - it should get raised when device enters Stop. Also, it is worth checking, that _all_ unused peripherals are switched off - clocks disabled in RCC->APB[12]ENR, example code only switches off SPI and radio.
(Sorry Miguel)

mluis1, 2020-06-09:
@altishchenko There is nothing to be sorry about.
It is additional information.

x893, 2020-06-09:
Easy way - use STK3200 or STK3700 from Silabs.

rebakker, 2020-06-10:
@mluis1, thank you for your reply. In the meantime I found the cause of the problem.

> The 2uA value is the value to be expected for MCU+Radio only.
> I don't know how you are measuring the power consumption. But, from the description I guess that you are also measuring the ST-Link, LED, ... power consumption.

As I described I disconnected **SB6** and applied the external +3.3V to **CN13**. So now this +3.3V is connected via **JP1**, **JP2** and **JP3** to **VDD_RF_LRA**, **VDD_USB_LRA** and **VDD_MCU_LRA**. These three power rails are only connected to the LoRa Module (Murata), it's decoupling capacitors and the unused user button B1.
But the +3.3V is also connected to R26 (4k7) in series with R27 (4k7), which is the cause for the extra current consumption.

> The best way to compute the power consumption is to use an oscilloscope where 2 probes are connected to each side of an 100 Ohm resistor in series with Murata module power supply.
> Then on the [...]

To measure the current flowing into CN13 I use a Joulescope JS110 (https://www.edn.com/tryout-joulescope-precision-dc-energy-analyzer/), which is a device which does all the described with a very high precision. Yesterday I even checked it's calibrations using a Krohn-Hite 523.