Lora-net/LoRaMac-node #908: UART RX not working in stop mode

rebakker, 2020-06-10:
When the LoRaMac-node is running on the B-L072Z-LRWAN1 and placed in sleep mode (using _BoardLowPowerHandler()_ with _UsbIsConnected = true;_ in _board.c_) it can still receive data via the UART. When it is placed in stop mode  (using _BoardLowPowerHandler()_ with _UsbIsConnected = false;_ in _board.c_), it is not receiving anything anymore.
There are several requirements to wake the UART from stop mode, as described in the ST application note AN4991. Have these requirement been implemented in the LoRaMac-node ? So changing the baudrate to something lower than the used 921600 will result in a wakeup of the MCU. 

Is this related to https://github.com/Lora-net/LoRaMac-node/issues/826?

altishchenko, 2020-06-11:
@rebakker Hello, the issue is unrelated. That problem stems from the one-at-a-time receive nature of the default HAL_USART driver that does a complete stop on any error like parity, overrun, framing, noise, etc. It is very picky.
In the current application configuration you can't wake up the MCU from the stop mode as described in the AN, because the default APB clocking is used for USART(2). To be able to wake up on receive from stop at least this should be changed in board.c:SystemClockConfig():
```
    PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_RTC|RCC_PERIPHCLK_USART2;
    PeriphClkInit.RTCClockSelection = RCC_RTCCLKSOURCE_LSE;
    PeriphClkInit->Usart2ClockSelection = RCC_USART2CLKSOURCE_HSI;
    if( HAL_RCCEx_PeriphCLKConfig( &PeriphClkInit ) != HAL_OK )
    {
        assert_param( FAIL );
    }
```
After that a few extra bits should be added to various places of board config functions - control bit setting in USART registers - UESM in CR1, WUS and WUFIE in CR3, etc.
Note: if you will try it, it is recommended to set UESM bit right before entry into Stop mode (BoardDeInitMcu() is a good place) and clearing this bit upon exit from Stop mode (BoardInitMcu()).

rebakker, 2020-06-11:
Thank you. I will take a look at it.

closed 2020-06-11 by rebakker