Lora-net/LoRaMac-node #910: Change default serial port for LOG Messages

gilbertoe, 2020-06-14:
Hi

I am developling a custom board based on B-L072Z-LRWAN1 reference design from ST. The reference design uses USART2 (PA2 and PA3 pins) for LOG messages. 

Is there a way to configure the LoraMac-node stack to use another serial port?  For instance the USART1 (PA10 and  PA9 pins). 

BR
Gilberto

altishchenko, 2020-06-15:
Hi, There is always a way :) It is not straightforward, but I can help you if you want. I'd suggest using e-mail for this as this is not really an "issue". Drop me a not at atishch@gmail.com and let me know the version of the library you are using.

gilbertoe, 2020-06-16:
> Hi, There is always a way :) It is not straightforward, but I can help you if you want. I'd suggest using e-mail for this as this is not really an "issue". Drop me a not at [atishch@gmail.com](mailto:atishch@gmail.com) and let me know the version of the library you are using.

Thanks. I sent you a message there.

closed 2020-06-16 by gilbertoe