Lora-net/LoRaMac-node #911: Any general documentation on editing Region files? (esp. AU915)

acutetech, 2020-06-15:
Is there any documentation on how the Region files are structured, and the changes that can/should be made, other than in these Gitgub posts?

I found that the RegionAU915.c & .h utilises radio channels that are not supported by actual gateways and networks. The Things Network says "Note that The Things Network uses 2nd Sub-Band only (channels 8 to 15 and 65). You’ll need to program the specific channels into the devices in order to make them work with TTN."

If I poke around these files, I find that I can restrict operation to the specified channels by making these changes to RegionAU915.c. Is this the correct fix?

> #define SUBBANDCHANNELS
> 
> #ifdef SUBBANDCHANNELS
>             NvmCtx.ChannelsDefaultMask[0] = 0xFF00;    // ch. 0-15
>             NvmCtx.ChannelsDefaultMask[1] = 0x0000;   
>             NvmCtx.ChannelsDefaultMask[2] = 0x0000;
>             NvmCtx.ChannelsDefaultMask[3] = 0x0000;
>             NvmCtx.ChannelsDefaultMask[4] = 0x0002;    // ch. 64-71
>             NvmCtx.ChannelsDefaultMask[5] = 0x0000;
> #else
>             NvmCtx.ChannelsDefaultMask[0] = 0xFFFF;
>             NvmCtx.ChannelsDefaultMask[1] = 0xFFFF;
>             NvmCtx.ChannelsDefaultMask[2] = 0xFFFF;
>             NvmCtx.ChannelsDefaultMask[3] = 0xFFFF;
>             NvmCtx.ChannelsDefaultMask[4] = 0x00FF;
>             NvmCtx.ChannelsDefaultMask[5] = 0x0000;
> #endif    // 2NDSUBBANDCHANNELS

and:

> #ifdef SUBBANDCHANNELS
> // TTN allows only one 500kHz channel (ch. 65)
> #define CHANNELS_MASK_500KHZ_MASK       0x0002
> #else
> // A mask to select only valid 500KHz channels (64-71)
> #define CHANNELS_MASK_500KHZ_MASK       0x00FF
> #endif    // SUBBANDCHANNELS

mluis1, 2020-06-16:
As per LoRaWAN specifications the AU915 region SHALL implement and use all 64 + 8 channels in order to be compliant.

This means that for this category of regions (AU915, US915) 64 + 8 channels gateways must be used.
Although in practice a lot of users are using 8 + 1 channels gateways.

On these regions the end-device must still use the 64 + 8 channels then it is up to the Network Server to setup the end-device with best channel mask at given time.

Thus, when using non compliant 8 + 1 channels gateways on these regions the users must accept that some *uplinks* will be lost up until the Network Server sets the right channel mask.
The Network servers have 2 possibilities to setup the channel mask. Either at the Join procedure by providing the `CFList` on `JoinAccept` message or by using the `LinkAdrReq` MAC commands.

This means that the region implementation files provided by this project aren't meant to be modified as they are compliant with LoRaWAN specifications where all 64 + 8 channels are to be used.

What can be done when operating on AU915 and US915 regions which are using non-compliant 8 + 1 channels gateways and we would like to speed up the Join procedure, is to setup the end-device to use a given channel mask.

Please find below how this can be achieved without breaking the LoRaWAN specification compliance as the end-device will revert to the 64 + 8 channels mask in case it looses connectivity with the Network server.

In the below example we enable the so called 1st 8 + 1 channels block.
In case another 8 + 1 channels block has to be used one has just to update ` uint16_t channelMask[]` variable initialization value.

- 1st block variable initialization:  `{ 0x00FF, 0x0000, 0x0000, 0x0000, 0x0001, 0x0000};`
- 2nd block variable initialization: `{ 0xFF00, 0x0000, 0x0000, 0x0000, 0x0002, 0x0000};`
- 3rd block variable initialization:  `{ 0x0000, 0x00FF, 0x0000, 0x0000, 0x0004, 0x0000};`
- 4th block variable initialization:  `{ 0x0000, 0xFF00, 0x0000, 0x0000, 0x0008, 0x0000};`
- 5th block variable initialization:  `{ 0x0000, 0x0000, 0x00FF, 0x0000, 0x0010, 0x0000};`
- 6th block variable initialization:  `{ 0x0000, 0x0000, 0xFF00, 0x0000, 0x0020, 0x0000};`
- 7th block variable initialization:  `{ 0x0000, 0x0000, 0x0000, 0x00FF, 0x0040, 0x0000};`
- 8th block variable initialization:  `{ 0x0000, 0x0000, 0x0000, 0xFF00, 0x0080, 0x0000};`

Example for NucleoL476 board: `src\apps\LoRaMac\periodic-uplink-lpp\NucleoL476\main.c`
```c
/*!
 * Main application entry point.
 */
int main( void )
{
...
    // The LoRa-Alliance Compliance protocol package should always be
    // initialized and activated.
    LmHandlerPackageRegister( PACKAGE_ID_COMPLIANCE, &LmhpComplianceParams );

// BEGIN - Workaround when using AU915/US915 with an 8 channels gateway setup
#if defined( REGION_AU915 ) || defined( REGION_US915 )
    MibRequestConfirm_t mibReq;
    // Enabling 1st block of 8 channels (0-7) + channel 64
    uint16_t channelMask[] = { 0x00FF, 0x0000, 0x0000, 0x0000, 0x0001, 0x0000};
    mibReq.Type = MIB_CHANNELS_MASK;
    mibReq.Param.ChannelsMask = channelMask;
    LoRaMacMibSetRequestConfirm( &mibReq );
    mibReq.Type = MIB_CHANNELS_DEFAULT_MASK;
    mibReq.Param.ChannelsDefaultMask = channelMask;
    LoRaMacMibSetRequestConfirm( &mibReq );
#endif
// END - Workaround when using AU915/US915 with an 8 channels gateway setup

    LmHandlerJoin( );

...
}
```

acutetech, 2020-06-16:
Thanks Miguel for the comprehensive explanation. I tried your fix on a variant of the LoRaMac Class A example. I print the channel masks and they are correct immediately prior to calling LoRaMacStart( ); but they have been changed back to the original by the time the code gets to JoinNetwork();

I think JoinNetwork() calls LoRaMacMlmeRequest() calls ResetMacParameters( ) calls RegionAU915InitDefaults() which puts back the original values.

I moved your code to immediately after the LoRaMacMlmeRequest() call - which has the feel of a nasty hack - but for some reason that is not working. So I am open to suggestions as to how your fix can be implemented more generally. Otherwise I am inclined to keep my own fix/hack.

I repeat my original request: is there any general documentation on the Region files and customisation for practical implementations? (I am finding another practical problem relating to AU915_DEFAULT_UPLINK_DWELL_TIME setting small payload sizes, that also looks like it needs a hack...)

mluis1, 2020-06-17:
It has been quite some time now that I didn't check AU915 implementation.
Currently, I have no access to a test network for AU915 region. As I need to setup one it will take some time before I can check it.

The region documentation is the Regional Parameters specification [v1.0.3revA](https://www.lora-alliance.org/resource-hub/lorawantm-regional-parameters-v103reva)

The region specific files shouldn't require any tweaks or changes as these files implement what is specified by the LoRaWAN specifications. This is true unless an issue is found.
Thus, it means that the provided examples should work as is.

Concerning the observed channel mask issue:
Recently I have been testing US915 region (similar to AU915) and it is operating as expected with this workaround for 8 + 1 channels gateway. Thus, I would say that AU915 region should work as expected as well.

Nevertheless I went trough the `RegionAU915.c` file and the only potential fix for the channel mask that I see for the moment is the one provided by the below patch.

**And again I would like to emphasis that this workaround for 8 + 1 channels gateway is not currently compliant with LoRaWAN specifications.**

```patch
diff --git a/src/mac/region/RegionAU915.c b/src/mac/region/RegionAU915.c
index 5d89c593..587784a6 100644
--- a/src/mac/region/RegionAU915.c
+++ b/src/mac/region/RegionAU915.c
@@ -847,9 +847,6 @@ int8_t RegionAU915AlternateDr( int8_t currentDr, AlternateDrType_t type )
 {
     static int8_t trialsCount = 0;
 
-    // Re-enable 500 kHz default channels
-    NvmCtx.ChannelsMask[4] = CHANNELS_MASK_500KHZ_MASK;
-
     if( ( trialsCount & 0x01 ) == 0x01 )
     {
         currentDr = DR_6;
```

Concerning the dwell time please read the below quote from the RP specification.  
You will notice that the end-devices shall start with an `UplinkDwellTime = 1` and then it is up to the network server to instruct the end-device to switch to `UplinkDwellTime = 0`  using **TxParamSetupReq** MAC command when required.

RP specification extract.
>The default JoinReq Data Rate is DR2 (SF10/125KHz), this setting ensures that end-devices are compatible with the 400ms dwell time limitation until the actual dwell time limit is notified to the end-device by the network server via the MAC command **TxParamSetupReq**.  
AU915-928 end-devices MUST consider UplinkDwellTime = 1 during boot stage until reception of the **TxParamSetupReq** command.  
AU915-928 end-devices MUST always consider DownlinkDwellTime = 0, since downlink channels use 500KHz bandwidth without any dwell time limit.

acutetech, 2020-06-18:
This is now somewhat off-topic, but TTN forum [has commented](https://www.thethingsnetwork.org/forum/t/txparamsetupreq-missing-with-au915/37886) on the UplinkDwellTime issue as follows: "It seems the above quote is from the 1.0.3 Regional Parameters. As far as I know, the TTN Community network is at 1.0.2, for which the 1.0.2 Rev B Regional Parameters don’t specify anything like that?"

Indeed, the two specs differ in this respect. There is also a discussion on AS923 dwell time [here](https://www.thethingsnetwork.org/forum/t/as923-dwell-time-limitation-for-australia/36601). 

In practise, I do not appear to receive a TxParamSetupReq message from my TTN server. Nor do I appear to receive any channel mask settings.

mluis1, 2020-09-01:
From LoRaWAN specification point of view the difference from 1.0.2 and 1.0.3 is the official addition of the Class B support plus the addition of the DeviceTimeReq MAC command.

On the other hand the differences between Regional Parameters specification 1.0.2revB and 1.0.3revA are mainly due to different countries regional regulations updates.

As you are aware TTN is currently working on their v3 stack update. The current v2 (TTN servers are using this version) implementation is not anymore being actively updated.

This project code base is compliant with LoRaWAN Specification v1.0.3 and on LoRaWAN Regional Parameters v1.0.3revA thus there is nothing more we can do on the subject.

So, to solve your current issue the possible solutions are:
- Change this project current AU915 implementation in order to be synced up with TTN v2 current implementation. (I think this is not advisable)
- Make a feature request on TTN v2 stack GitHub repository in order for them to support 1.0.3revA regional parameters specification. Most certainly they will not work on such improvement.
- Wait for TTN to be updated to v3 stack.
- Move your devices to TTN v3 stack. 
   - TTI provides hosted v3 stack access as a service.
   - The v3 stack source code is available on GitHub and you can install it on your own network while waiting for TTN servers update.

If you don't mind I close this issue.

closed 2020-09-01 by mluis1