Lora-net/LoRaMac-node #916: Confused about porting this to an Arduino library

ZaneL, 2020-06-24:
Hello,

I'm using v4.4.4 and I'm trying to port this code over to the Arduino platform for the SX1262. It seems to be going okay, but I'm a little confused about the RTC board-specific functions. Really all I need is the Class A functionality -- but it seems that in LoRaMAC.c, class B and class A code are all tightly coupled so it requires you to implement the RTC functionality, even when it isn't necessary for class A.

I've determined through trial and error that these are the functions that I must implement. What do you recommend I do for the RTC functions if I have no RTC on-board? Can I just leave the functions empty and it will work okay for Class A functionality? Or is there some easy way to implement these RTC functions using the built-in Arduino millis() function? 

Even if I were to connect an RTC and I were to use a library like Time/TimeLib.h, I'm at a complete loss as to how I would use the functions from that TimeLib library to implement these Loramac-node RTC functions. It seems overly complex (when looking at rtc-board.c as an example).

[https://pastebin.com/YB8RiMZp](https://pastebin.com/YB8RiMZp)

ZaneL, 2020-06-24:
Maybe something like this? Can I just use milliseconds as ticks?

...and then in the RtcSetAlarm function I just setup a timer that will call TimerIrqHandler( );

Does that sound correct?

[https://pastebin.com/gju2tLBu](https://pastebin.com/gju2tLBu)

altishchenko, 2020-06-24:
Sure, there is no strict requirement for the proper real time. Monotonic time counter is quite enough. Requirements begin with Class B and other things which can be safely ignored in Class A. Be warned though, this library is pretty heavy when compiled and might not fit in your board.

ZaneL, 2020-06-25:
Sounds good, thank you!

closed 2020-06-25 by ZaneL