Lora-net/LoRaMac-node #924: sx1278 enter standby when RX1 have error(classC)

wwx0715, 2020-07-02:
Using 4.4.4, and work at classC.

Server send downlink packet every 10 seconds, and deviceresponse a uplink packet.

The problem is that the device do not receive downlink packet any more until the device send a uplink packet.
After test 12 hours, the problem may occur about 4 times.

Normally, for classC, after RX1 window timeout, mcu will receive rx timeout interrupt at dio1 and  sx1276 enter continuous rx mode before RX2 window.
I add some log info and to debug this problem, and found that dio0 timeout interrupt at RX1 window may not be receive by mcu,  and RX2 window will not be setted for MacCtx.RxSlot is RX_SLOT_WIN_1 at OnRxWindow2TimerEvent. And then SX1276OnTimeoutIrq is enterred (SX1276.Settings.State is still RF_RX_RUNNING, and OpenContinuousRxCWindow will fail when call RegionRxConfig, and then the problem will not be recovered until an uplink packet).

Actually, sx1278 is at standby mode(I print the mode reg period), so, is it possible that sx1278 not trigger timeout interrupt or mcu lost this interrupt? and how to recover?
Thanks.

wwx0715, 2020-07-02:
read sx1278 register REG_LR_IRQFLAGS when SX1276OnTimeoutIrq enter, it is 0x80 (RFLR_IRQFLAGS_RXTIMEOUT). It means that timeout is really ocured, but interupt is not triggered for mcu.

wwx0715, 2020-07-06:
After uplink package, device enter RX C->RX1->RX C->RX2->RX C, at the first RX C, at the begin of function SX1276SetRx, IRTFLAGS is 0, at  the end( before call SX1276SetOpMode), IRTFLAGS  is 0x80, and at between of them, SX1276OnDio1Irq is enter, but IRTFLAGS  is 0 at its handle. any idea?

wwx0715, 2020-07-08:
Last update, the abnormal dio1 interrupt occoured at SX1276SetRx to set REG_LR_IRQFLAGSMASK(LoRa). IRTFLAGS  is 0x00 printed before  setting REG_LR_IRQFLAGSMASK, is 0x80 printed after  setting REG_LR_IRQFLAGSMASK(LoRa), and at dio1 interrupt handle IRTFLAGS  is 0. The problem can be recovered after next uplink, and is not occured frequently, so it maybe not affect application, but the problem is really exist.

mluis1, 2022-01-12:
If you don't mind I convert this issue to a discussion topic. Maybe someone can help you with the observed issue.

It has to be noted that the v4.4.4 version is quite old now and we would recommend to updated your code base to the latest version as several issues have been addressed and may solve your observed issue.

From our own tests we are not able to reproduce the issue which makes it difficult to help you.