Lora-net/LoRaMac-node #926: SX126xSetTxContinuousWave not updating internal state tracker

andysan, 2020-07-05:
When transmitting a CW using `SX126xSetTxContinuousWave`, the internal operating mode state tracker isn't updated. It seems like a call to 'SX126xSetOperatingMode` is missing. This would become an issue if the operating mode is used to control an RF switch.

mluis1, 2020-07-06:
Thanks for the finding.

The following patch should solve the issue. Please let me know if you are OK with it.

```patch
 src/radio/sx126x/sx126x.c | 2 ++
 1 file changed, 2 insertions(+)

diff --git a/src/radio/sx126x/sx126x.c b/src/radio/sx126x/sx126x.c
index 5f99056e..2dfd4db0 100644
--- a/src/radio/sx126x/sx126x.c
+++ b/src/radio/sx126x/sx126x.c
@@ -341,11 +341,13 @@ void SX126xSetCad( void )
 void SX126xSetTxContinuousWave( void )
 {
     SX126xWriteCommand( RADIO_SET_TXCONTINUOUSWAVE, 0, 0 );
+    SX126xSetOperatingMode( MODE_TX );
 }
 
 void SX126xSetTxInfinitePreamble( void )
 {
     SX126xWriteCommand( RADIO_SET_TXCONTINUOUSPREAMBLE, 0, 0 );
+    SX126xSetOperatingMode( MODE_TX );
 }
 
 void SX126xSetStopRxTimerOnPreambleDetect( bool enable )
```

andysan, 2020-07-06:
> Thanks for the finding.
> 
> The following patch should solve the issue. Please let me know if you are OK with it.

Thanks for the quick turnaround! The should work, it's very similar to the patch I have been using. It might be a good idea to update the state before sending the `RADIO_SET_TXCONTINUOUSWAVE` command though since I need to use the state tracker to control the RF switch (I suspect it would be Very Bad to dump a CW into an improperly configured switch).

mluis1, 2020-07-06:
Thanks for the feedback.

I used that order because all other operating mode control functions were doing this in the same order.
I will update all other functions according to your proposition.

closed 2020-07-06 by mluis1