Lora-net/LoRaMac-node #928: SX126x FSK bandwidth looks wrong

jamesl-dm, 2020-07-10:
RadioSetRxConfig is called with the FSK _half _sided__ bandwidth set to 50 kHz, and the AFC bandwidth set to 83 kHz. But the SX126x driver seems to look up the bandwidth in a table of _double _sided__ bandwidths. So I think it is configuring the radio for a 50 kHz double sided bandwidth instead of the intended 100 kHz bandwidth.

The SX126x datasheet also says:

> The bandwidth must be chosen so that Bandwidth[DSB] ≥ BR + 2*frequency deviation + frequency error

So it looks like the SX126x has no provision for narrowing the bandwidth after an AFC phase. 
Maybe the driver should be using the AFC bandwidth at all times?
2x 83 kHz = 166 kHz, so we have a choice of 156 kHz at register setting 0x1A, or 187 kHz at 0x12.

I haven't tested any of this yet - just going by the datasheet.

closed 2020-09-01 by mluis1

mluis1, 2020-09-01:
Indeed the provided bandwidth has to be doubled. The proposed fix does this now.

Concerning the AFC bandwidth the SX126x radios automatically set the required parameters to perform the AFC based on the normal operation provided bandwidth.

jamesl-dm, 2020-09-01:
Shouldn't the << 1 be inside the bracket in that commit? It looks like it is doubling the register value.

mluis1, 2020-09-01:
Yes, thanks for the finding. I'll provide an update in the coming minutes.

mluis1, 2020-09-01:
Fix done. Thanks again.