Lora-net/LoRaMac-node #929: Duty cycle issue on v4.4.4

flyingmaster, 2020-07-13:
I am using LoRaMac v4.4.4 and working on EU868 with duty cycle on.  
According to LoRaWAN sepc 1.0.3，the join tansmit time should  be less than 36sec in the first hour after power-up or reset.
The TOA of the join request on DR0 is 1.483sec, so there should only transmit 24 (i.e. 36/1.483) join request in the first hour.
But currently I can sent 45 join requests in the first hour:
  1) In the first 150 seconds,  24 join requests can be sent out continously
  2) The 25th join request will hit duty cycle restriction. 
  3) from then on, every 150 seconds,  the next join request can be sent successfully.

I found in UpdateTimeCredits()， the TimeCredits will be recovered along with the LastBandUpdateTime, which will increase the 
next transmit opportunity. 
```
    // Get the difference between now and the last update
    band->TimeCredits += TimerGetElapsedTime( band->LastBandUpdateTime );
```
 The same problem can be easily reproduced during uplink message sending after join.   In the first hour after power up or the hour after keeping one hour slience, the total tansmit time of uplink message will violate ducty cycle restriction a lot.

djaeckle, 2020-08-04:
Hi @flyingmaster 

thanks for the report. We can confirm the issue. In general, this issue is only related to the backoff algorithm of the join requests. We are already working on a solution.

mluis1, 2020-09-29:
The commit 0efc1a7 solves this issue. Unfortunately, the commit message has been wrongly labeled and it should have been "Issue #929 - Fixed duty-cycle implementation".

The new way of computing the duty-cycle brings some drawbacks as well as improvements. The drawbacks are quite tricky to solve and in the end it is a question of compromise.

The improvements are that one can perform burst transmissions before getting restricted.

The drawbacks are quite tricky to explain but, I'll try to do it in the following paragraphs.
We will assume a `1%` sub-band duty-cycle restriction.

As you know the new algorithm is based on a `1hour` sliding window where credits are consumed.
This means that we will have `36000ms` (`36seconds`) of credits available per hour for a `1%` duty-cycle restriction.
We have 2 different situations to consider:  
1.	The end-device transmits uplinks periodically.
2.	The end-device transmits a few uplinks and then sleeps for a long period and repeats the process.

In case 1 we will end up with a `2 %` duty-cycle restriction in the 1st hour of operation (until the algorithm stabilizes) and `1%` for the subsequent hours of operation.  
The ETSI regulations aren't very clear if the duty-cycle restriction applies to all channels under the sub-band or if it applies per channel inside the sub-band.  
In current implementation we apply the duty-cycle restriction to all channels under the sub-band.
This means that if the ETSI applies the restriction per channel then the `2%` duty-cycle restriction at algorithm initialization could be OK as for instance the 3 default LoRaWAN channels are located under the same sub-band (we end in a reduced duty-cycle usage per channel).  
In case the ETSI applies the restriction per sub-band then we could reduce the amount of initial credits by 2 (`18000ms`) which would give us a `1%` duty-cycle restriction in the 1st hour of operation and `0.5%` the subsequent hours of operation.
In order to use the halved time credits one can change the observation period from `3600000` to `1800000` by providing the following definition to the compiler `-DDUTY_CYCLE_TIME_PERIOD=1800000`  

In case 2 we will most certainly only have a `2%` duty-cycle restriction as there is no way (at least at our current knowledge) to detect when the algorithm re-initializes.  
In this case either the application ensures by design that the `1%` limit is never reached or one can also provide the compiler time definition `-DDUTY_CYCLE_TIME_PERIOD=1800000`

Hope the explanation is clear enough.

mluis1, 2020-10-13:
In order to ensure that duty-cycle restrictions are respected we decided to change the observation time period default value to `1800000`.

closed 2020-10-13 by mluis1

flyingmaster, 2020-10-14:
Thanks for your workaroud method.  It can solve the issue with the data uplink duty cycle from 0.5% to 1%.
But I notice the join request  duty cycle is fixed to 0.5%, which results in the observation time period is 1800s while the DCWaitime is 1 hour.
Could you differentiate the observation time period by 3600s for unjoined case and 1800s for joined case in SetMaxTimeCredits()？