Lora-net/LoRaMac-node #932: Stripped down version of LoRaMAC-node for class A devices

ZaneL, 2020-07-15:
Would it be possible to provide a stripped down version of this library for simple class A devices. No NVM, no RTC, no class B support, etc. I've mostly ported this to Arduino (a teensy) and it's pretty ridiculous how many files are needed to make this library build. It seems that 90% of the code is totally unnecessary for my use case -- all I want to do is have the device join a network over OTAA and then send packets...

The only example I can find where someone attempted to do this is here: https://github.com/HelTecAutomation/ASR650x-Arduino/tree/master/cores/asr650x/loramac

You can see that they stripped out all of the NVM+RTC related calls and most of the files in the mac folder are gone. I'm not even sure how they decided what could be removed, it seems like a daunting task.

cjhdev, 2020-07-27:
Short of someone providing a nice worked example, I think your best bet is to look at and copy how other projects integrate it. 

There's the one you've linked. I've also seen integrations in mynewt-core and ArduinoCore-stm32l0. The former is barely recognisable if you are comparing to the default project structure.

LoRaMAC-node has a verbose style and I can't imagine that will change in future releases. Most integrators will probably leave the project as-is in a vendor directory so that updates can be applied easily.

There are alternatives that have a more compact style but they aren't maintained as well as LoRaMAC-node.

mluis1, 2020-10-21:
Can we close this issue?

mluis1, 2020-11-26:
No feedback we close the issue

closed 2020-11-26 by mluis1