Lora-net/LoRaMac-node #941: How to know when is necessary to execute LoRaMacProcess function?

lucaspeixotot, 2020-07-31:
I'm using the LoRaMac stack on my board from Silicon Labs and I'm using MicriumOS as my RTOS. In MicriumOS I can enter in low power mode just when there are no tasks ready to run. Thus, my LoRa task needs to be waiting for something to my board enter in low power, it can't be running based on some time delay.

In examples from LoRaMac node, for the stack works is necessary that the LoRaMacProcess() and Radio.IrqProcess() functions are running in a loop when the board is awake. So I would like to run the loop just when being necessary to execute either LoRaMacProcess() or Radio.IrqProcess(). Radio.IrqProcess() is easy to know when being necessary to run. However, how can I know when I need to execute LoRaMacProcess())? 

Whether I know it, I could use a semaphore to wait for either an interruption in DIO1 to run Radio.IrqProcess() function or something else to run LoRaMacProcess() function.

MarekNovakACRIOS, 2020-07-31:
Hello Lucas,
To my understanding of the stack, the events can be only triggered based on external interrupts (covered in Radio.IrqProcess()) and based on expiration of a timer (check timer.c file).
Therefore, what you need is a task/thread, which is run when a timer expires and executes the LoRaMacProcess(), also when and IRQ happens, you need to call Radio.IrqProcess()) and then LoRaMacProcess(), in this order.
Please, check also this file - https://github.com/Lora-net/LoRaMac-node/blob/master/src/boards/B-L072Z-LRWAN1/rtc-board.c, which is an implementation of RTC wrapper for a board based on STM32 - it is used to set timeout before the next software timer in the queue expires, then it triggers and interrupt:
```
void HAL_RTC_AlarmAEventCallback( RTC_HandleTypeDef *hrtc )
{
    TimerIrqHandler( );
}
```
After this, you should also unblock your task/thread checking the stack... 

Hope I helped you....

lucaspeixotot, 2020-08-03:
Thanks for your help my friend! Now everything is working as I want, thanks again!

closed 2020-08-03 by lucaspeixotot