Lora-net/LoRaMac-node #942: Why USE_LRWAN_1_1_X_CRYPTO is disabled and RJ_COUNT_0, and RJ_COUNT_1 aren't declared?

khariya, 2020-08-04:
Hello,

I got a JOIN_FAILED error on a LoRa node even when the server sends a JOIN_ACCEPT packet.
After some digging, it seems the `USE_LRWAN_1_1_X_CRYPTO` should be enabled which is disabled by default.
I enabled it and OTAA was successful.

I can't find any documentation on what this definition does. Can someone explain or point me to a document and why is it disabled by default?

On enabling the definition, the compiler throws error on `RJ_COUNT_0` and `RJ_COUNT_1` undeclared. This was a simple fix though.

mluis1, 2020-09-01:
Form your description we don't know which version of the LoRaWAN specification you are using neither the expected version on the network server side.

We suppose that you are using the code provided on master branch or a tagged release.
As mentioned on the [README.md](https://github.com/Lora-net/LoRaMac-node/blob/master/README.md) file the master branch or tagged versions are compliant with LoRaWAN 1.0.3 specification and this is why the `USE_LRWAN_1_1_X_CRYPTO ` is disabled.

By disabling the `USE_LRWAN_1_1_X_CRYPTO ` by default for 1.0.x versions reduces the final code size as the 1.1.x features aren't required.

Long time ago we decided to have only one implementation for the cryptographic tasks in order to ease the development of the different versions of the LoRaWAN specification. By doing so we are trying to avoid as much as possible to have divergent code bases.

The `RJ_COUNT_0` and `RJ_COUNT_1`  are only defined by LoRaWAN 1.1.x specifications. This means that they are only defined by the 1.1.x code base which can be currently found on [feature/5.0.0](https://github.com/Lora-net/LoRaMac-node/blob/feature/5.0.0) branch.
Please note that there are plenty of other things that are changed between 1.0.x and 1.1.x versions of the specification.

From your description you said that the Join procedure is successful when you enable `USE_LRWAN_1_1_X_CRYPTO ` so, I guess that the network server you are trying to connect to is using 1.1.x version of the specification and your end-device is setup as 1.1.x device while actually it is a 1.0.x device.

The end-device and network server LoRaWAN versions must match in order for them to work as expected.
That said you have 2 options:
1. Continue to use this project master branch and change the end-device network server settings to use a 1.0.x version of the specification.
2. Use this project feature/5.0.0 branch.

closed 2020-09-01 by mluis1

khariya, 2020-09-01:
Thanks for the detailed explanation. I'm sorry that I assumed USE_LRWAN_1_1_X_CRYPTO would only be available in the code base that supports LoRaWan 1.1.x.

My code is based on the develop branch because the Readme.md says there are two active branches master and develop.

I do understand the difference between the LoRaWan 1.0.x and 1.1.x and the importance of version match on the node, and the network server.

I'm planning to support only LoRaWan 1.1.x, so I chose to use the develop branch.

Are you suggesting I should switch to the (inactive) feature/5.0.0 branch?

mluis1, 2020-09-01:
Currently develop branch only provides a LoRaWAN 1.0.4 specification preview. (The 1.0.4 specification hasn't yet been released)

The feature/5.0.0 is active and being updated. If your goal is to use 1.1.x version of the specification then indeed this is the branch to use.

Concerning the readme.md file it was our intention to merge the feature/5.0.0 branch into develop. But, this will only happen after the 1,0,4 LoRaWAN specification will be released. So, in the mean time please use feature/5.0.0 branch base for your 1.1.x developments.

khariya, 2020-09-01:
Ok. Thank you!