Lora-net/LoRaMac-node #951: Wrong FCnt in gateway

ImanIDG, 2020-08-15:
Hi there. I'm using LoRaMac on ST32L073RZ and SX1276 (EU868 Region). I tried to modify this code for my purpose but now i face an interesting problem. At transmission, I cant see my node on Server. when i checked, the problem was wrong fcnt and devaddr reading in gateway. 
you can see the mac frame sent to sx1276. I set 0x11111111 as devaddr but i get 1111400D in gateway. also instead of fcnt = 1, i get 32785 (0x8011) in gateway. it seems the tx frame got shifted for two bytes when sx1276 tries to send it but i cant find the source of this issue. I have checked preamble length but it's ok.

[image]

mklil, 2020-08-17:
in case you did modify the spi init  code in (spi-board.c), check if you have the right clk phase and polarity settings.

mluis1, 2020-09-01:
We close the issue then.

closed 2020-09-01 by mluis1