Lora-net/LoRaMac-node #954: How to receive downlink packets without sending uplink packets in class C?

lucaspeixotot, 2020-08-24:
Hello, I made a porting of LoRaMac-node, and everything is working well, except by not receiving downlink packets async when I'm in class C. I have a device, and I'm setting all of its keys and making the join process. After that, I'm able to send uplink and receive downlink packets immediately after uplink is sent. However, I guess that in class C, I could receive downlink packets even without sending uplink packets. 

My network server is in EveryNet, and I'm using its DataAPI to send a downlink_claim. I supposed that I should be receiving the downlink packet in McpsIndication callback after the downlink_claim. The fact is that I'm not receiving the packet.

Do I need to make something different to receive downlink packets even without send uplink packets in class C? Should it be working?

x893, 2020-08-24:
Interesting question
But рow does the gateway know the communication parameters?
Telephaty ? Last session ? Fixed ?
Can't say about EveryNet but Semtech packet-forwarder not support.

lucaspeixotot, 2020-08-24:
I don't know exactly how the gateway knows the communication parameters, but I think that this is not the problem.
[image]
This is the diagram provided by EveryNet about the downlink_claim operation. All the steps above are being executed by the network server and application server. However, when the network server sends the "downlink", this packet is not being received in my end-device. I think that I don't need to know and set the communication parameters, apparently, this is already done by the network server, isn't a manual operation. However, this is just my opinion, can be wrong.

I was hoping that when the network server sent the downlink, my McpsIndication callback could be triggered, but as it is not, I don't know what is the problem exactly.

x893, 2020-08-25:
I can't say about Eve... but you can try semtech opensource GW and see all trace info abot up/down link. Semtech GW codes not a complex and easy to debug and change as you wants.

nestorayuso, 2020-08-25:
The node must perform at least one uplink, so the network server learns your nearest gateway to make subsequent downlinks

lucaspeixotot, 2020-08-25:
@nestorayuso I took care of it. I'm always sending one uplink packet, it is being received and after that, I'm trying to send downlink_claims without receiving uplink packets. This image below shows that the communication between the network server and the application server is correct. First, there is a downlink request, so the network server process it and sends the downlink packet.
[image]

Apparently, in my code, everything is working either. The stack is opening the rx2 window continually, with datarate 8 and frequency 923.3MHz, so I think that I should be receiving the downlink packets.

In the image below you can see the variables that I can configure when I'm starting the application server. The 'network' field could be a problem, but I don't think so. In the network server, apparently, all the uplinks that I'm sending and the downlinks that I'm receiving have the same network id. So I configure my application server with it.

[image]

Is there some configuration that I should be doing to use the end-device in class C?

adixmasz, 2020-08-27:
In my case (LoraMac-node-4.4.2 and ChirpStack Server) it seems to works fine.
A node is able to gets multiple downlink packets  (unconfirmed) after the first uplink.

mluis1, 2020-09-01:
Have you setup the end-device on the network server as a Class C enabled device?

I never used Everynet network server but, I would assume that there is an option indicating that the end-device is a Class C end-device.
Without such indication the network server will only schedule downlinks on the Rx1 or Rx2 windows.

As stated in previous comments the Class C end-devices are working with other network servers.

Also, I would verify that I didn't missed something while doing the project porting.
Something that may help is by providing `USE_RADIO_DEBUG` definition to the compiler. When this symbol is enable 2 GPIOs are used to indicate the current radio operating mode. 1 pin is used to indicate when the Radio is in Tx mode and the other when it is in Rx mode. When both are set to 0 the Radio is in sleep mode.
Connecting these 2 pins to an oscilloscope you should see something similar to the below drawings

Class A
[image]

Class C:
[image]

lucaspeixotot, 2020-09-02:
> In my case (LoraMac-node-4.4.2 and ChirpStack Server) it seems to works fine.
> A node is able to gets multiple downlink packets (unconfirmed) after the first uplink.
> `###### ===== MLME-Confirm ==== ######
> STATUS : OK
> ###### ===== JOINED ====
> 
> OTAA
> 
> DevAddr : 3153CE6D
> 
> DATA RATE : DR_0
> ###### ===== MCPS-Confirm ====
> 
> STATUS : OK
> ###### ===== UPLINK FRAME 1 ====
> 
> CLASS : C
> 
> TX PORT : 2
> TX DATA : CONFIRMED - ACK
> 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10
> 1;32m1B 1C 1D 1E 1F 20
> 21 22 23 24 25 26 27
> ###### ===== MCPS-Indication ====
> 
> STATUS : OK
> ###### ===== DOWNLINK FRAME 1 ====
> 
> RX WINDOW : C
> RX PORT : 2
> RX DATA :
> 4B 55 50 41
> 
> DATA RATE : DR_0
> RX RSSI : -59
> RX SNR : 8
> ###### ===== MCPS-Indication ====
> 
> STATUS : OK
> ###### ===== DOWNLINK FRAME 2 ====
> 
> RX WINDOW : C
> RX PORT : 2
> RX DATA :
> 4B 55 50 41
> 
> DATA RATE : DR_0
> RX RSSI : -60
> RX SNR : 7
> `

Interesting, I'm using the LoRaMac 4.4.4, but if your code is working in 4.4.2 should not be a problem works in 4.4.4.

lucaspeixotot, 2020-09-02:
> Have you setup the end-device on the network server as a Class C enabled device?
>
> I never used Everynet network server but, I would assume that there is an option indicating that the end-device is a [...]

I didn't know about this USE_RADIO_DEBUG flag. I will configure it to see what is happening.

About this issue, I was thinking and I realized that: I'm using AU915 and when I'm sending an uplink, it is sent in DR_2 = SF10/125KHz. Hence the downlink sent after uplink by the network server is in 925.7MHz, SF100/500KHz. However, when I'm sending a downlink without a uplink, the network server is always sending in 923.3MHZ and SF12/500KHz. 

Apparently this channel and SF that the network server is using is the default configuration for RxCwindow in AU915. May this be the problem?

lucaspeixotot, 2020-09-02:
@mluis1 you made a commit 16 days ago named "Back port 1.0.4 region implementation improvements". There are changes in the region files. There is some relevant modification for AU915? May I need to update my code to have these improvements?

mluis1, 2020-10-21:
It is always better to update the project to the latest version when possible. 
The latest version brings bug fixes as well as improvements.

Concerning the specific issue reported here I don't think that "Back port 1.0.4 region implementation improvements" will fix your observed issue with Everynet network server.

The RxC window parameters are defined by the regional parameters specification and are the same as for Rx2 window. In case of AU915 region the RxC window parameters are:
>•	The RX2 (second receive window) settings uses a fixed data rate and frequency. Default parameters are 923.3MHz / DR8

closed 2020-10-21 by mluis1