Lora-net/LoRaMac-node #956: EU868 DutyCycle Restriction v4.4.4 class A end device

dennisvbussel, 2020-08-27:
Hi,

As hardware configuraiton I have a custom board implementation of a nRF58232 in cobination with the SX1262 Radio.
As software I use the LoRaMac-node 4.4.4 release. Using the EU868 Region parameters. LoRa class A end device.

The stack implementation works fine (Joining, uplink/downlink) except for the duty cycle limitation. I can send as much data as i want and never get the LORAMAC_STATUS_DUTYCYCLE_RESTRICTED.

My test procedure:
1. Send unconfirmed LoRa message containing 20 bytes of random data on port 2 to The Things Network (TTN)
2. When i get the MCPS_UNCONFIRMED in the function McpsConfirm I execute step 1 again.
With the RTT debugger is connected I can observer the MacCtx in "realtime".

Observations:
- MacCtx.DutyCycleWaitTime is always 0.
- MacCtx.NvmCtx.LastTxDoneTime increases with every TX
- MacCtx.NvmCtx.LastTxDoneTimeOff is always 0.

At this point I'm clueless. Is there anybody that can shine some light on this? 

Thanks in advance,
Dennis

DoubleCheeseBacon, 2020-08-28:
Hi,

Like you I was a bit surprise at first by this behavior but, as I have understood, now the duty-cycle is seen like a credit of time on air.

So, until you have used all your credits you can send frames continuously.
After that, you will have to wait a certain amount of time to restore enough credits.

Have a look at : 
- RegionCommon.c/RegionCommonUpdateBandTimeOff() ==> DEBUG_TRACE("i : %d, %d, %d\r\n", i, bands[i].TimeCredits, creditCosts);

Credits are in [ms].

Hope it helps ...

Arnaud

mluis1, 2020-09-01:
@DoubleCheeseBacon Is right. Recently we have modified the way the duty cycle is restricted. 
Please refer to issue #520 for further information.

Please also note that we are currently fixing the issue reported on issue #929 where the restriction happens too late.
Up until #929 is fixed the work around is to not transmit uplinks too often.

closed 2020-09-01 by mluis1