Lora-net/LoRaMac-node #959: Persistent context storage management

gilbertoe, 2020-09-07:
Hi

I am trying to avoid to resend the Join message every time the node is restarted. Is there any information on the minimum information that needs to be stored in order to preserve all the necessary keys? The default method provided in the examples is not suitable  due to the reduced number of write cycles that stm32l0's eeprom support.

BR
Gilberto

MedadRufus, 2020-09-07:
With a wear leveling algorithm, you should be able to write to EEPROM well
over 100k cycles

Medad

> Hi
>
> I am trying to avoid to resend the Join message every time the node is
restarted. Is there any information on the minimum information that needs
to be stored in order to preserve all the necessary keys? The default
method provided in the examples is not suitable due to the reduced number
of write cycles that stm32l0's eeprom support.
>
> BR
> Gilberto
https://ci5.googleusercontent.com/proxy/U70GviQwN92-U1qQ30-qZq3ZErG0IaDl5JaUoBtlIJ2PMd6KjQfMRsLk0MfYotcQRVY7kecpgsPUnnb-TejP_kvaa9v8yyi8eNs62QGl4Mfpiq9MI4zElL55Qn12tfAllwPqvPJj3a5bMumDFFvJsk9GiBdaYKwVl-yLiMffGud_hgsJDiQCToTS6cobDVbnrDg5D804RBwh9IzGqY7AYYxagaE=s0-d-e1-ft#https://github.com/notifications/beacon/AGMSV4L7G6QHVUGX2Q2D2EDSEQSOBA5CNFSM4Q5EAKW2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4KLGSPFQ.gif>

gilbertoe, 2020-09-09:
Hi

I have an additional question related with the same topic:

What do Crypto and SecureElement store when the basic level of Context Storage is enable?  Do they have some  effect on the Join procedure?

BR
Gilberto

mluis1, 2020-11-26:
@gilbertoe Yes, Crypto and SecureElement have effect on the Join procedure.

We just provide an example on how the non-volatile data can be persisted.
How one can reduce the number of writes cycles or how to handle the memory wear leveling is out of scope of this project.

Please find below some clues on how this kind of problem could be solved:
- Use an external EEPROM, Ferroelectric RAM (F-RAM), or Magnetoresistive RAM (MRAM) chips. Usually the number of write cycles is bigger than internal MCU FLASH memories. [wikipedia](https://en.wikipedia.org/wiki/Non-volatile_memory)
- Reduce the number of times the NVM is updated. Only write the NVM every 10 `NvmContextChange` notification. The only drawback happens if the device is powered off between actual NVM writes.
- Implement your own wear leveling algorithm.

closed 2020-11-26 by mluis1