Lora-net/LoRaMac-node #960: Region selection/hopping during runtime?

AndreiHanu, 2020-09-13:
Hi,

We are currently developing a GPS tracker based on the Murata CMWX1ZZABZ chip and need to be able to select the LoRa region/frequency at runtime based on the GPS location and a preset geofence.

Is it possible to implement that feature using this stack? 

The initialization code on this page (http://stackforce.github.io/LoRaMac-doc/_q_u_i_c_k__s_t_a_r_t__g_u_i_d_e.html) seems to indicate that you can do that.

`static void McpsConfirm( McpsConfirm_t *mcpsConfirm )
{
  // Implementation of the MCPS-Confirm primitive
}
 
static void McpsIndication( McpsIndication_t *mcpsIndication )
{
  // Implementation of the MCPS-Indication primitive
}
 
static void MlmeConfirm( MlmeConfirm_t *mlmeConfirm )
{
  // Implementation of the MLME-Confirm primitive
}
 
static void MlmeIndication( MlmeIndication_t *mlmeIndication )
{
  // Implementation of the MLME-Indication primitive
}
 
static void OnMacProcessNotify( void )
{
  // Mac notification. Process run function
}
 
LoRaMacPrimitives_t LoRaMacPrimitives;
LoRaMacCallback_t LoRaMacCallbacks;
LoRaMacStatus_t Status;
 
int main( void )
{
  LoRaMacPrimitives.MacMcpsConfirm = McpsConfirm;
  LoRaMacPrimitives.MacMcpsIndication = McpsIndication;
  LoRaMacPrimitives.MacMlmeConfirm = MlmeConfirm;
  LoRaMacPrimitives.MacMlmeIndication = MlmeIndication;
  LoRaMacCallbacks.GetBatteryLevel = BoardGetBatteryLevel;
  LoRaMacCallbacks.GetTemperatureLevel = NULL; // apply board specific temperature reading
  LoRaMacCallbacks.NvmContextChange = NvmCtxMgmtEvent;
  LoRaMacCallbacks.MacProcessNotify = OnMacProcessNotify;
  **// Initialization for the region EU868
  Status = LoRaMacInitialization( &LoRaMacPrimitives, &LoRaMacCallbacks, LORAMAC_REGION_EU868 );**
 
  if( Status == LORAMAC_STATUS_OK )
  {
    // Initialization successful
  }
}`

However, the build instructions in the README.md file seems to indicate the region is selected during compilation and cannot be changed at runtime.

Can you provide some clarification on this?

altishchenko, 2020-09-14:
@AndreiHanu Hello, you can run `LoRaMacInitialization` anytime you like. It is not necessarily done at startup, but depending on your application's needs. We do it in our project whenever we need it. Just don't forget to call `LoRaMacDeInitialization`, before you restart the LORA stack with the new regional settings.

AndreiHanu, 2020-09-14:
Thank you Alex!

closed 2020-10-21 by mluis1