Lora-net/LoRaMac-node #963: Timer Possible Issue

LocatorCJSC, 2020-09-17:
I am having an issue when starting a new Timer with a Timeout value smaller than the Timer in the head.

Can you please confirm or deny the issue by running this PoC code?

```C
static TimerEvent_t t1;
static TimerEvent_t t2;

static void OnT1( void* context )
{
    return;
}

static void OnT2( void* context )
{
    return;
}

int main( void )
{
    // Configure t1 to fire in 10000ms
    TimerInit( &t1, OnT1 );
    TimerSetValue( &t1, 10000 );

    // Configure t2 to fire in 2000ms
    TimerInit( &t2, OnT2 );
    TimerSetValue( &t2, 2000 );

    // Start t1
    TimerStart( &t1 );

    // Wait 5000 before starting t2
    DelayMs(5000);

    // Start t2
    TimerStart( &t2 );

    uint32_t t2_started_at = ( uint32_t ) TimerGetCurrentTime( );
    uint32_t t1_expected_fire_at = 5000; // 10000 - 5000
    uint32_t t2_expected_fire_at = 2000;

    bool t1_fired = false;
    bool t2_fired = false;

    while ( 1 )
    {
        uint32_t time_passed;

        time_passed = TimerGetCurrentTime( ) - t2_started_at;

        if ( t1_fired == false && TimerIsStarted( &t1 ) == false )
        {
            t1_fired = true;
            printf("t1 fired in %" PRIu32 ", expected was %" PRIu32 "\n", time_passed, t1_expected_fire_at);
        }
        if ( t2_fired == false && TimerIsStarted( &t2 ) == false )
        {
            t2_fired = true;
            printf("t2 fired in %" PRIu32 ", expected was %" PRIu32 "\n", time_passed, t2_expected_fire_at);
        }
    }
}
```

My result is:

```
t2 fired in 6998, expected was 2000
t1 fired in 7004, expected was 5000
```

LocatorCJSC, 2020-09-17:
This might be the reason: in the `TimerStart` function the elapsed time (in this case the 5000ms of the delay) is being added to the timeout value of the newly started Timer:

```C
        elapsedTime = RtcGetTimerElapsedTime( );
        obj->Timestamp += elapsedTime;

        if( obj->Timestamp < TimerListHead->Timestamp )
        {
            TimerInsertNewHeadTimer( obj );
        }
        else
        {
            TimerInsertTimer( obj );
        }
```

So the next alarm is being set at 2000 + 5000 = 7000 instead of just 2000.

LocatorCJSC, 2020-09-18:
This seems to fix the problem for me:

```C
void TimerStart( TimerEvent_t *obj )
{
    uint32_t elapsedTime = 0;

    CRITICAL_SECTION_BEGIN( );

    if( ( obj == NULL ) || ( TimerExists( obj ) == true ) )
    {
        CRITICAL_SECTION_END( );
        return;
    }

    obj->Timestamp = obj->ReloadValue;
    obj->IsStarted = true;
    obj->IsNext2Expire = false;

    if( TimerListHead == NULL )
    {
        RtcSetTimerContext( );
        // Inserts a timer at time now + obj->Timestamp
        TimerInsertNewHeadTimer( obj );
    }
    else
    {
        elapsedTime = RtcGetTimerElapsedTime( );

        if( obj->Timestamp + elapsedTime < TimerListHead->Timestamp )
        {
            TimerEvent_t* cur;
            uint32_t old =  RtcGetTimerContext( );
            uint32_t now =  RtcSetTimerContext( );
            uint32_t deltaContext = now - old; // intentional wrap around

            for( cur = TimerListHead; cur != NULL; cur = cur->Next )
            {
                if( cur->Timestamp > deltaContext )
                {
                    cur->Timestamp -= deltaContext;
                }
                else
                {
                    cur->Timestamp = 0;
                }
            }

            // Inserts a timer at time now + obj->Timestamp
            TimerInsertNewHeadTimer( obj );
        }
        else
        {
            obj->Timestamp += elapsedTime;
            TimerInsertTimer( obj );
        }
    }
    CRITICAL_SECTION_END( );
}
```

altishchenko, 2020-09-18:
Hello, I think you somewhat misunderstand the mechanics behind. Adding elapsed time is the correct way to start the new alarm when another one is or was started. I have this timer code running for two years now without any issues on STM32L051.

Your initial code does not contain RTC initialization(!) was this intentional? Try calling RtcInit() before your experiments.

Also I'd recommend checking the times by two global **volatile** variables, set in the corresponding timer handler OnT1, OnT2 to TimerGetCurrentTime().
Like in:
```
volatile uint32_t t1_fired = 0;

void OnT1(void *context)
{
    t1_fired = TimerGetCurrentTime();
}
```

LocatorCJSC, 2020-09-22:
Hello, thank you.

Yes, I do call `BoardInitMcu( );` and `BoardInitPeriph( );` in my code, I forgot to include it in the cut-down version posted here.

> I have this timer code running for two years now without any issues on STM32L051.

That's good to know, I'm using another microprocessor (not included in this repository) so there is a good chance that the RTC module's port is doing something wrong.

> Adding elapsed time is the correct way to start the new alarm when another one is or was started.

I think I understand the mechanism, I left that "adding elapsed time" part when the new timer is going to be fired later than the earliest exiting. But when the newly added timer is going to become the next one to fire, so a new alarm will be set with its "timeout" value counting from now on, does it need the elapsed time to be added to the "timeout" value?

altishchenko, 2020-09-22:
@LocatorCJSC Make sure that the functions ticks2ms and ms2tick do the right thing, proper functioning of RTC is essential here. We had serious issues with one custom board where RTC balance SYNC/ASYNC prediv and base frequency were not as expected by the code in this repository. Basically code here expects 32768 Hz xtal, with 1 second = **1024** timer ticks.

> But when the newly added timer is going to become the next one to fire, so a new alarm will be set with its "timeout" value counting from now on, does it need the elapsed time to be added to the "timeout" value?

Yes it does, have a close look at the TimerIrqHandler(), especially at the deltaContext variable and its usage.

Basically all this means "Set the new alarm in *this many* ticks **since the last alarm was started**", so when RtcSetAlarm calculates the exact time for the new alarm it considers the time passed *since* the last Alarm was set (elapsedTime) and takes that into account calculating the new value. Otherwise setting the new alarm will require re-initializing timestamps of all the other previously queued timers.

Trust me, it plays out correctly in the end, we have alarms as short as 10-20 ms firing exactly and at the proper times for the correct custom HW management.

LocatorCJSC, 2020-09-22:
> so when RtcSetAlarm calculates the exact time for the new alarm it considers the time passed since the last Alarm was set (elapsedTime) and takes that into account calculating the new value

That was exactly what was wrong in my RTC port. Short alarms are being emulated in this port (because alarm's granularity is one minute) using a tick interrupt, and in that case the elapsed time wasn't being taken into account.

> Otherwise setting the new alarm will require re-initializing timestamps of all the other previously queued timers

And that was exactly how I "fixed" it in my code posted above.

Thank you!

closed 2020-09-22 by LocatorCJSC