Lora-net/LoRaMac-node #964: Issue when running new certification requierements for EU868-870 (v1.6)

Pirakutu, 2020-09-25:
Hi everyone,

I found a problem on the stack when running the test 2.5.11 Incorrect MAC Commands of the new testing specs. The test sends different incorrect MAC commmands to the DUT and expects a specific behaviour. The problem im getting is connected to the LinkADRReq. This command is sent a bunch of times (once with params out of spec, once as incomplete MAC frame and so on). I get a problem after having gotten this command with the false length (incomplete frame). The test spec expects the DUT to not react to it, my device (the stack) sends an answer though. I am using the stack version 4.4.3, do you happen to know this phenomena?

Cheers,

Andres

ConnyBusy, 2020-09-25:
@Pirakutu,

This issue is fixed in stack version 4.4.4

Pirakutu, 2020-09-25:
I see the same issue with v4.4.4, the stack rejects the false commands, which in my opinion is an alright behavior, the problem is that the testing spec expexts the DUT to ignore the false command, so no rejection allowed.

ConnyBusy, 2020-09-25:
which command is supposed to be ignored??

Pirakutu, 2020-09-28:
For example the command -> 0x03010000 (incomplete LinkADRReq). According to the test specifications, this command MUST be ignored. The stack doesnt ignore it, it rejects it (it sends a LinkADRAns != OK), which in my opinion should also be a valid answer, but because the test specs are defined that way, the test doesnt pass. There is a sort of mechanism inside the stack that only adds an answer if the number of processed bytes is 5, but the problem is that the functions called before dont really care that only 4 bytes were received and set the number of processed bytes automatically to 5, which makes the stack send an answer (once again, the answer is a rejection or !=OK, which is actually good, meaning that the stack doesnt take invalid parameters).

ConnyBusy, 2020-09-28:
I don't think that this should happen in 4.4.4 the code below prevents that. 

`
 // Make sure to parse only complete MAC commands

        if( ( LoRaMacCommandsGetCmdSize( payload[macIndex] ) + macIndex ) > commandsSize )

        {

            return;

        }
`

Pirakutu, 2020-09-28:
You are right, just checked that function, that should actually do the job, thanks! I will try it out on my device.

Pirakutu, 2020-09-30:
The stack 4.4.4 does the job, thanks

closed 2020-09-30 by Pirakutu