Lora-net/LoRaMac-node #965: Problem with FSK50 maximum payload test

Pirakutu, 2020-09-30:
Hi guys,

I am having problems passing the FSK50 maximum payload test on the new EU868 specs with the stack v4.4.4, is this problem known to any of you? Thanks!

Cheers,

Andres

Pirakutu, 2020-09-30:
Another question, shall i go with the master or with the release version? ive seen there have been some fixes made on the master branch, how far checked are these fixes?

mluis1, 2020-12-14:
Could you please check if commit f8d1950619d0d1f2c835d5df3eb719cbcb85a3e9 solves your issue?

We noticed that using FifoEmpty IRQ was too much constraint when the MCU was using low power modes.
In order to solve that we changed the IRQ to FifoLevel instead.

mluis1, 2020-12-14:
Master branch is most of the time stable. Prior to push to GitHub we usually perform some internal tests.

In may happen time to time that it gets not stable. Most of the time it happens when we accept pull-requests on GitHub.

Pirakutu, 2020-12-23:
Thanks, will test this and come back to you (it will probably be next year as we went ahead and certified the device without this option, it would still be interessting to have it working though)

mluis1, 2021-03-03:
Had you some time to perform your checks?
Can we close this issue?

mluis1, 2021-03-18:
No feedback, we close the issue.

closed 2021-03-18 by mluis1