Lora-net/LoRaMac-node #971: how dose this project build and debug?

enddls, 2020-10-21:
hello

I have keil compliler so I wnat to complile and debug this project at keil.

how dose this project build and debug?

please help me

mluis1, 2020-10-21:
Please find attached a zipped Keil project example for NucleoL073+SX1261MBXBAS platform.

[periodic-uplink-lpp.zip](https://github.com/Lora-net/LoRaMac-node/files/5414008/periodic-uplink-lpp.zip)

You need to create the following directories and place the extracted file there.
`keil\NucleoL073\loramac\periodic-uplink-lpp`

If you have another platform then you need to adapt the project for it.
please create the following directories `keil\<your platform name>\loramac\periodic-uplink-lpp`, copy the attached unzipped Keil project file and adapt it to your platform.

MedadRufus, 2020-10-21:
Many thanks for the NucleoL073 example. Is there an example for B-L072Z-LRWAN1 as well? It will make life a lot easier to use the loramac-node code if it works right out of the box with Keil or other known IDEs, instead of figuring out how to use the current workflow which takes quite a while to setup. I still have not figured out how to fully debug using the workflow required by this project and hence I use the I-CUBE-LRWAN from STMicroelectronics. I would indeed prefer to use the Loramac-node code if it were easier to use.

enddls, 2020-10-21:
thank you. Miguel Luis

I want to create NucleoL476RG+LR1110MB1DJS project.

but I can't find that.

so I create project LoRaMac-node-master.

but i don't know to create project.

mluis1, 2020-10-21:
As you may know we are very resource limited and we cannot support every IDE that exists.
We used to maintain Keil projects on this repository but, a lot of work was required to maintain them up to date and to validate them.

Some time ago we decided to only support CMAKE/GCC/GDB based projects as they are supported by the major operating systems (Windows, Linux, MacOS).

That, said please find the requested examples.

B-L072Z-LRWAN1 project file to be placed under `keil\B-L072Z-LRWAN1\loramac\periodic-uplink-lpp`
[periodic-uplink-lpp.zip](https://github.com/Lora-net/LoRaMac-node/files/5414731/periodic-uplink-lpp.zip)

NucleoL476+LR1110MB1XXS project file to be placed under `keil\NucleoL476\loramac\periodic-uplink-lpp`
[periodic-uplink-lpp.zip](https://github.com/Lora-net/LoRaMac-node/files/5415587/periodic-uplink-lpp.zip)

I didn't checked if produced binaries are working on real hardware. I have just verified that each target successfully compiles.

MedadRufus, 2020-10-21:
Thank you thank you thank you @mluis1. This will allow me to test run the code from this repo on keil, check everything is working, and use it as a reference when eventually using the CMAKE/GCC/GDB workflow.

Thanks again.

enddls, 2020-10-22:
thank you @mluis1.

I want to communicate Lr1110 board and gateway which is P-NUCLEO-LRWAN2 gateway board.

P-NUCLEO-LRWAN2 gateway board supports CLASS : A and frequency : EU-868.

I need to Lr1110  class A program. 

So can I just change the main.c file "apps\LoRaMac\periodic-uplink-lpp\NucleoL476\main.c" to "\apps\LoRaMac\classA\NucleoL476\main.c"  in your project?

mluis1, 2020-10-22:
The `classA`, `classB` and `classC` examples will be deprecated in the near future.

The `periodic-uplink-lpp` is the replacement example for the previous 3 examples. By default this example is setup to act as Class A end-device.  
In order to change the active class one can change  `LORAWAN_DEFAULT_CLASS` definition.  
With this example an end-device always starts as a ClassA end-device. Once connected to a network server it will switch to the class specified by the  `LORAWAN_DEFAULT_CLASS` definition if necessary.

main.c file:
```c
/*!
 * LoRaWAN default end-device class
 */
#define LORAWAN_DEFAULT_CLASS                       CLASS_A
```

*I would strongly recommend that you base your application on the `periodic-uplink-lpp` example as classA, classB and classC examples will be removed in near future.*

In case you still want to use the `classA` example, yes, your proposed change is the right one.

enddls, 2020-10-22:
thank you @mluis1.

periodic-uplink-lpp include LoRaWAN "class C type"?

my final program will set LoRaWAN "class C" and include "class A", WIFI_SCAN and GNSS SCAN.

honestly, I dont know LoRaWan program so I ask you the LoRaWan program.

thank you for your help.

mluis1, 2020-10-22:
As it looks like you want to use your end-device in Class C mode then you need to change `LORAWAN_DEFAULT_CLASS` definition to `CLASS_C`.

Once the device is connected to the network server it will automatically switch to Class C.

main.c file:
```c
/*!
 * LoRaWAN default end-device class
 */
#define LORAWAN_DEFAULT_CLASS                       CLASS_C
```

As per LoRaWAN specifications an end-device is always a ClassA device (Tx->Rx1->Rx2). This is also true when an end-device is operating in ClassB or ClassC modes.  
The ClassB just opens periodic reception windows while not transmitting (ClassA: Tx->Rx1->Rx2).  
The ClassC just sets the radio in continuous reception mode while not transmitting(ClassA: Tx->Rx1->Rx2).  

Please note that:
Class A end-devices are the most low-power
Class B end-devices are more power consuming than Class A as periodic reception windows are opened
Class C end-devices are the most power angry as they are most of the time in reception mode.

I would recommend you to read the LoRaWAN specifications in order to better understand how everything works.

Another point you need to be aware of is that this repository only provides LoRaWAN usage examples. Which means that for LR1110 radio we only use it as a transceiver.
This repository does not implement the required functionality to handle WIFI_SCAN and GNSS_SCAN features. It means that you will need to add these features implementation by your self.

As you intend to use the WIFI_SCAN and GNSS_SCAN features it will be maybe easier for you to use the following LR1110 supporting software repositories.
https://github.com/Lora-net/lr1110_evk_demo_app
https://github.com/Lora-net/lora_edge_tracker_ref_design
https://github.com/Lora-net/lr1110_modem_application_example

As we are diverging from the initial request I will close this issue.

In case you require further support I would recommend that you either contact the Semtech support team or that you post your questions on [LoRa Developper Portal](https://lora-developers.semtech.com/knowledge-base/forum/)

closed 2020-10-22 by mluis1

enddls, 2020-10-22:
> As it looks like you want to use your end-device in Class C mode then you need to change `LORAWAN_DEFAULT_CLASS` definition to `CLASS_C`.
>
> Once the device is connected to the network server it will [...]

@mluis1, thank you for your help.