Lora-net/LoRaMac-node #972: Downlink not received at SF = 7

kevinaldianw, 2020-10-21:
I'm using two boards. The first one is B-L072Z-LRWAN1. The second one is NucleoL073 + SX1276MB1LAS. I'm using Class A project with ADR on. I tried to send a confirmed message with both boards and they did not received any downlink at SF=7 and SF=8. The downlink is received at SF=9.  Is there any solution for this problem? Thank you

cjhdev, 2020-10-21:
Sounds like a timing problem. Are you using an RC oscillator for timing?

The RC oscillator on those targets can cause problems if:

1. It isn't calibrated to the expected nominal frequency
2. The +/- uncertainty hasn't been compensated for by the library

At SF7 the device needs to open a window no smaller than ~8ms to receive at least ~5ms of preamble. You get +/-1.5ms of slack. Increasing SF increases the period of each symbol (doubles?), hence why it works with higher SF.

That said, maybe your timing is perfect and the network just isn't sending downlinks.

kevinaldianw, 2020-10-21:
I'm using the HSI clock for both boards. This problem did not occur with other boards with HSE clock.

kevinaldianw, 2020-10-30:
This problem is solved by using DMA for UART communication

wisang1999, 2020-11-03:
I have the same issue here. I just test the downlink and confirmed packet at SF7 & SF8 on AS923 freq plan but no downlink/ack received.

kevinaldianw, 2020-11-03:
This problem only happens when I use v4.4.5. It does not happen when I use v4.4.4

mluis1, 2020-11-26:
Can we close this issue?

kevinaldianw, 2020-11-26:
Sure. v4.4.4 works fine

closed 2020-11-26 by kevinaldianw

sslupsky, 2021-05-21:
@kevinaldianw I do not understand, you closed this issue when the later release has the reported issue?  Does that mean the solution is to downgrade the firmware to an earlier version?

nilotnas, 2021-06-04:
I'm facing the same issue with a STM32L073 + SX1276 custom board on v4.4.7. Anyone else?