Lora-net/LoRaMac-node #979: Can't activate on RU864 region

Xeenych, 2020-10-25:
I have wap LoRa8 Kit gateway.
I'm using a custom board similar to Skim880b.

When region is set to EU868 (on node and on gateway) OTAA activation completes succesfully.

When region is set to RU864, board can't activate.

On the gateway I see join request, and then join accept. RSSI is -60dB, SNR is 10dB.
Board sometimes shows Rx2 Timeout and sometimes Rx2 Error, which is caused by CRC error internally from Sx1272 chip interrupt.

altishchenko, 2020-10-25:
Does your gateway/network server configuration match the RU864 configuration of the library? There are two RU864 config variants I know of. Make sure that your network server/gateway uses 869.1 MHz as an RX2 frequency with full DR range,

Xeenych, 2020-10-25:
I don't know.
There is mikrotik router OS v6.46.1 and only single RU864 option

altishchenko, 2020-10-25:
And what is the network server in use? (you can drop me a message at atishch @ gmail.com as this issue looks more of a network config problem then library issue).

Xeenych, 2020-10-25:
i'm using router.eu.thethings.network
I've sent you a email message

Xeenych, 2020-10-26:
It is strange but
switching to digitalcatapult-uk-router (ttn.thethingsconnected.net) solved the problem

closed 2020-10-26 by Xeenych