Lora-net/LoRaMac-node #980: how do LR1110 connect https://eu1.loriot.io/ sever?

enddls, 2020-10-28:
hi, 

I test "loriot sever( mode : Packet Forwarder STM) ↔ P-NUCLEO-LRWAN2(gateway) ↔ LR1110 board with class A program"

gateway's teminal is shown log that is recieved LR1110 class A data

but loriot sever is not shown data.

I have tried to solve this problem for three days.... but I can't solve.

how can Lr1110 board(class A program) communicate loriot sever?

please help me.

x893, 2020-10-28:
You can use wareshark and see netwotk activity from gateway to loriot server.

enddls, 2020-10-28:
thank you x893

when to enroll deivce at loriot server input deviceeui, appeui, appkey value. but class A source appkey no input value.

so when DeviceState is DEVICE_STATE_START, i programed As follows : 

uint8_t u8t_devEui[8] = LORAWAN_DEVICE_EUI;
mibReq.Type = MIB_DEV_EUI;    // MIB_DEV_EUI is loriot enroll deive eui
mibReq.Param.DevEui = u8t_devEui;
LoRaMacMibSetRequestConfirm( &mibReq );

uint8_t u8t_joinEui[8] = LORAWAN_JOIN_EUI;
mibReq.Type = MIB_JOIN_EUI;  // MIB_JOIN_EUI is loriot enroll join eui
mibReq.Param.JoinEui = u8t_joinEui;
LoRaMacMibSetRequestConfirm( &mibReq );

uint8_t u8t_sePin[4] = SECURE_ELEMENT_PIN;
mibReq.Type = MIB_SE_PIN;      // MIB_SE_PIN is "0000"
mibReq.Param.JoinEui = u8t_sePin;
LoRaMacMibSetRequestConfirm( &mibReq );

mibReq.Type = MIB_APP_KEY;
uint8_t AppKey[16] = LORAWAN_APP_KEY;   // LORAWAN_APP_KEY is loriot enroll appkey
mibReq.Param.AppKey = AppKey;
LoRaMacMibSetRequestConfirm( &mibReq );

mibReq.Type = MIB_NWK_KEY;
uint8_t NwkKey[16] = LORAWAN_NWK_KEY;  //LORAWAN_NWK_KEY is loriot enroll appkey value
mibReq.Param.NwkKey = NwkKey;
LoRaMacMibSetRequestConfirm( &mibReq );

Is that right?

(SX1272MB2) example source "end device" dosen't exist sepin.

but LoRaMac-node class A source at lr1110 exist sepin. 

so sepin setting value is "0x00 0x00 0x00 0x00".

Is that right?

I don't know how to close the difference between SX1272MB2 example source and LoRaMac-node class A source.

I want to LR1110 communicate loriot sever.

help me.

I installed wireshark, but I do not know how to use it, so I will study and check it.

enddls, 2020-10-29:
> You can use wireshark and see netwotk activity from gateway to loriot server.
thanks to x893.

I search wire shark setup. but wire shark can capture lan card of computer.

is it right?

gateway(P-NUCLEO-LRWAN2) is direct connect LAN line. 

therefore wire shark cann't capture gateway(P-NUCLEO-LRWAN2) send data.

closed 2020-11-02 by enddls