Lora-net/LoRaMac-node #984: how does LR1110 communicate https://eu1.loriot.io/ server(packet forward)?

enddls, 2020-11-02:
hi,

I test "loriot sever( mode : Packet Forwarder STM) ↔ P-NUCLEO-LRWAN2(gateway) ↔ LR1110 board with class A program"

lr1110 joined loriot.io server. and lr1110 is showed DevAddr and DATA RATE.

lr1110 send data to gateway. gateway recieve data. so gateway's teminal is shown log that is recieved LR1110 data.

but loriot sever is not shown recieve data.

I have tried to solve this problem for weeks.... but I can't solve.

how can Lr1110 board(class A program) communicate loriot sever?

please help me.

/////gateway log//////
LRX: 00D89EDFA99AEFC7825673502D565250080F0F38EAD51B
JUL: {"rxpk":[{"tmst":2545443628,"chan":5,"rfch":1,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":-8.0,"rssi":-118,"size":23,"data":"ANie36ma78eCVnNQLVZSUAgPDzjq1Rs="}]}
JDL: {"txpk":{"brd":0,"ant":0,"tmst":2550443628,"freq":868.1,"rfch":0,"powe":14,"modu":"LORA","ipol":true,"datr":"SF12BW125","size":33,"data":"IM20FMQOIhxPYKpVFAcCd0mvNLrMOosnd+B8naJGrxzp","imme":false,"codr":"4/5"}}
LTX: 20CDB414C40E221C4F60AA551407027749AF34BACC3A8B2777E07C9DA246AF1CE9
LTS: 2546017555 2550443628 4426073
LRX: 40228F2F30800100021FC6CFEB09D1FC01613082EC61FEA8D55B416D67
JUL: {"rxpk":[{"tmst":2554289188,"chan":4,"rfch":0,"freq":867.900000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":-13.5,"rssi":-116,"size":29,"data":"QCKPLzCAAQACH8bP6wnR/AFhMILsYf6o1VtBbWc="}]}
LRX: 40228F2F3080020002FB83CA4B212F73D0C9F79C55F25510B144D05DEE
JUL: {"rxpk":[{"tmst":2559558500,"chan":1,"rfch":0,"freq":867.300000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":-9.3,"rssi":-117,"size":29,"data":"QCKPLzCAAgAC+4PKSyEvc9DJ95xV8lUQsUTQXe4="}]}
LRX: 40228F2F308003000214CFAB93BF3F71E456B2A231C21D5180C38EAD8D
JUL: {"rxpk":[{"tmst":2563602964,"chan":7,"rfch":1,"freq":868.500000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":-9.0,"rssi":-116,"size":29,"data":"QCKPLzCAAwACFM+rk78/ceRWsqIxwh1RgMOOrY0="}]}

mluis1, 2020-11-02:
@enddls Why creating a new issue and closing the previous one (https://github.com/Lora-net/LoRaMac-node/issues/980) on the same subject?

Some elements that must be taken in account:
1. Which activation method is to be used? (OTAA, ABP)
2. Will the Semtech Join Server cloud service be used?
3. From where the required EUI and keys are generated
4. Which secure-element implementation will be used.

The easiest and faster way is to use the `SOFT_SE` implementation.  
Once you successfully connect to loriot.io using the `SOFT_SE` you can try to use the `LR1110_SE` implementation.
Please read the following chapter [lr1110-se](https://github.com/Lora-net/LoRaMac-node#lr1110-se) on the [README.md](https://github.com/Lora-net/LoRaMac-node/blob/master/README.md) file for further details on how to use the `LR1110_SE` implementation.

To use the `SOFT_SE` implementation just edit `src\peripherals\soft-se\se-identity.h` file with your EUI and Keys.

In order for loriot.io to decode and list the received data all identity elements must correspond.

enddls, 2020-11-02:
hi, @mluis1.

Why creating a new issue and closing the previous one (#980) on the same subject?
- add test conditions. (join loriot server so loriot server is changed DevAddr)
- there is no answer.

1. Which activation method is to be used? (OTAA, ABP)
- setting OTAA. 
```
#define OVER_THE_AIR_ACTIVATION									1
/*!
 * When using ABP activation the MAC layer must know in advance to which server
 * version it will be connected.
 */
#define ABP_ACTIVATION_LRWAN_VERSION_V10x						0x01000300 // 1.0.3.0

#define ABP_ACTIVATION_LRWAN_VERSION							ABP_ACTIVATION_LRWAN_VERSION_V10x

/*!
 * Indicates if the end-device is to be connected to a private or public network
 */
#define LORAWAN_PUBLIC_NETWORK									true

/*!
 * Current network ID
 */
#define LORAWAN_NETWORK_ID										( uint32_t ) 0

```

2. Will the Semtech Join Server cloud service be used?
- tested only loriot server.
- after this problem and almanac solve, i try to join semtech server. 

3. From where the required EUI and keys are generated
-when DeviceState is DEVICE_STATE_START, I set APPKEY.
```
	case DEVICE_STATE_RESTORE:
		{
			// Try to restore from NVM and query the mac if possible.
			if( NvmCtxMgmtRestore( ) == NVMCTXMGMT_STATUS_SUCCESS )
			{
#ifdef PRINT_ENABLE
				printf( "\n###### ===== CTXS RESTORED ==== ######\n\n" );
#endif
			}
			else
			{
				// Read secure-element DEV_EUI, JOI_EUI and SE_PIN values.
				mibReq.Type = MIB_DEV_EUI;
				LoRaMacMibGetRequestConfirm( &mibReq );
				memcpy1( devEui, mibReq.Param.DevEui, 8 );
#ifdef PRINT_ENABLE
			printf( "\n###### ===== DEVICE_STATE_RESTORE ==== ######\n\n" );
			printf( "DevEui      : %02X", mibReq.Param.DevEui[0] );
			for( int i = 1; i < 8; i++ )
			{
				printf( "-%02X", mibReq.Param.DevEui[i] );
			}
			printf( "\n" );
#endif
				mibReq.Type = MIB_JOIN_EUI;
				LoRaMacMibGetRequestConfirm( &mibReq );
				memcpy1( joinEui, mibReq.Param.JoinEui, 8 );
#ifdef PRINT_ENABLE		
			printf( "JoinEui     : %02X", mibReq.Param.JoinEui[0] );
			for( int i = 1; i < 8; i++ )
			{
				printf( "-%02X", mibReq.Param.JoinEui[i] );
			}
[... 136 lines cut ...]
			printf( "\nABP\n\n" );
			printf( "DevAddr     : %08lX\n", DevAddr );
			printf( "\n\n" );
		#endif
			mibReq.Type = MIB_NETWORK_ACTIVATION;
			mibReq.Param.NetworkActivation = ACTIVATION_TYPE_ABP;
			LoRaMacMibSetRequestConfirm( &mibReq );

			DeviceState = DEVICE_STATE_SEND;
#else
			JoinNetwork( );
#endif
		break;
		}
```

4. Which secure-element implementation will be used.
- I used lr1110-se. 
- i will try to use soft-se with your opinion.
- can lr1110 use soft-se?

but sx1272MB2(endnode project) successfully connect and communicate loriot server.
- this project use soft-se.

thank you your help.

enddls, 2020-11-02:
thank you, @mluis1.

lr1110 use soft-se. successfully connect and communicate loriot server.

how do I make lr1110-se setting equal to soft-se setting?

mluis1, 2020-11-02:
To use `LR1110_SE` Please read the following chapter [lr1110-se](https://github.com/Lora-net/LoRaMac-node#lr1110-se) on the [README.md](https://github.com/Lora-net/LoRaMac-node/blob/master/README.md) file for further details on how to use the `LR1110_SE` implementation.

By reading the previous README.md chapter you will notice that you need to write the required DevEUI, JoinEUI, AppKey to the radio prior to run the examples. This is something that should be done while you manufacture your end-devices. This has to be done only once per end-device.

First I would recommend that you use the `periodic-uplink-lpp` example as a base for your firmware. `classA`, `classB` and `classC` are deprecated and will be removed on next release.

In the patch provided at following link (please remove ".txt" extension after downloading the patch)
[lr1110-se-commissioning.patch](https://github.com/Lora-net/LoRaMac-node/files/5475462/lr1110-se-commissioning.patch.txt)
I have created a small application which writes the required elements to the LR1110 crypto engine.

Prior to compile the example 
1. Please update `LORAWAN_DEVICE_EUI`, `LORAWAN_JOIN_EUI`,  `LORAWAN_APP_KEY` and `LORAWAN_NWK_KEY`
2. Compile the code
3. Program the MCU
4. Once the program has finished the required elements have been updated.
5. Change `.vscode/settings.josn` to compile `periodic-uplink-lpp`
6. Compile the `periodic-uplink-lpp` example.
7. Program the MCU
8. The end-device should now be using the keys you provided on step 1.

Please note that I did not run or tested the provided example. I just compiled it.

altishchenko, 2020-11-02:
@mluis1 Miguel, there is a typo in the patch - you use `SecureElementSetDevEui()` twice - where it should be and a few lines lower where `SecureElementSetJoinEui()` should go. Beware!

mluis1, 2020-11-02:
@altishchenko Thanks for the finding.

Please find below the updated patch.
[lr1110-se-commissioning.patch](https://github.com/Lora-net/LoRaMac-node/files/5475888/lr1110-se-commissioning.patch.txt)

enddls, 2020-11-03:
thanks, @mluis1 

I will check "lr1110-se-commissioning.patch" and "README.md"

closed 2020-12-09 by mluis1