Lora-net/LoRaMac-node #987: SX1272 SNR Values

prabwyld, 2020-11-11:
As per the datasheet the SNR value obtained is denoted in the following way:
SNR dB  = PacketSnr [two s complement]/4 . I am unable to correlate this to the implementation in the code which is as follows:

SX1272.Settings.LoRaPacketHandler.SnrValue = ( ( ( int8_t )SX1272Read( REG_LR_PKTSNRVALUE ) ) + 2 ) >> 2;

Is this the correct implementation? How is adding 2 to the register giving us a 2's compliment?

mluis1, 2020-11-11:
Yes, the implementation is correct.

The radio `REG_LR_PKTSNRVALUE` is already a 2's complement value. 
The `SX1272Read` function returns an `uint8_t` variable type. So, in order to have a 2's complement value the returned value is cast to `int8_t`.

The `+2` is there to perform the rounding operation after the division as `2/4=0.5`

prabwyld, 2020-11-11:
Thank you for the clarification. I was assuming that the raw value from the register has to be converted first to 2's compliment. That clears my doubt and also +2 addition as well.
-Cheers.

closed 2020-11-11 by prabwyld