Lora-net/LoRaMac-node #989: Compiling the project

JchenEng, 2020-11-23:
Hello,

I fairly inexperienced with the whole cmake software, but essentially I am encountering some issues compiling the project. I have followed the development environment (https://github.com/Lora-net/LoRaMac-node/blob/master/doc/development-environment.md). I am trying to use the CMake GUI (CMake 3.19.0), but I receive the following errors:

Please specify the TOOLCHAIN_PREFIX !
 For example: -DTOOLCHAIN_PREFIX="C:/Program Files/GNU Tools ARM Embedded" 
The C compiler identification is unknown
The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:18 (project):
  The CMAKE_C_COMPILER:

    /bin/arm-none-eabi-gcc.exe

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:18 (project):
  The CMAKE_CXX_COMPILER:

    /bin/arm-none-eabi-g++.exe

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

Where is the source code: "C:/Users/XXXXX/Desktop/Electronics/LoRaMac-node-master"
Where to build the binaries: "C:/Users/XXXXX/Desktop/Electronics/build"

I have tried adding a new entry called DTOOLCHAIN_PREFIX to where I installed the GNU tools ARM Embedded 
[image]
but that did not resolve my issue.

djaeckle, 2020-11-25:
Hi @JchenEng ,

could you please double check that you have defined every symbol which is available in the file ".vscode/settings.json"? E.g. have you defined `-DCMAKE_TOOLCHAIN_FILE=cmake/toolchain-arm-none-eabi.cmake`?

In general, i would recommend to use VS Code which helps to have an easy start.

mluis1, 2020-11-26:
The README.md file provides the instructions on how to build the project at chapter [Building Process](https://github.com/Lora-net/LoRaMac-node#building-process)

The commands example were provided to build periodic-uplink-lpp example for NucleoL476 platform with LR1110MB1DIS MBED shield and using LR1110 pre-provisioned secure-element.

This means that you need to adapt the command according to your needs.

```bash
$ mkdir build
$ cd build
$ cmake -DCMAKE_BUILD_TYPE=Release \
        -DTOOLCHAIN_PREFIX=""C:/9 2019-q4-major"" \
        -DCMAKE_TOOLCHAIN_FILE="../cmake/toolchain-arm-none-eabi.cmake" \
        -DAPPLICATION="LoRaMac" \
        -DSUB_PROJECT="periodic-uplink-lpp" \
        -DCLASSB_ENABLED="ON" \
        -DACTIVE_REGION="LORAMAC_REGION_EU868" \
        -DREGION_EU868="ON" \
        -DREGION_US915="OFF" \
        -DREGION_CN779="OFF" \
        -DREGION_EU433="OFF" \
        -DREGION_AU915="OFF" \
        -DREGION_AS923="OFF" \
        -DREGION_CN470="OFF" \
        -DREGION_KR920="OFF" \
        -DREGION_IN865="OFF" \
        -DREGION_RU864="OFF" \
        -DBOARD="<YOUR PLATFORM NAME>" \
        -DMBED_RADIO_SHIELD="<YOUR SHIELD NAME IF NECESSARY>" \
        -DSECURE_ELEMENT="SOFT_SE" \
        -DSECURE_ELEMENT_PRE_PROVISIONED="ON" \
        -DUSE_RADIO_DEBUG="ON" ..
$ make
```

JchenEng, 2020-11-28:
Hey everyone, 

Sorry for the delay. I am using a NucleoL073 board with the radio shield (CWX1ZZABZ) which I believe is a SX1276MB1MAS. 
@mluis1 I tried following the Building Process, but it returns the following error message: 
```
CMake Error at src/CMakeLists.txt:72 (include):
  include could not find load file:

    stm32l0

CMake Error at src/apps/LoRaMac/CMakeLists.txt:171 (include):
  include could not find load file:

    gdb-helper

CMake Error at src/apps/LoRaMac/CMakeLists.txt:172 (include):
  include could not find load file:

    binutils-arm-none-eabi

CMake Error at src/apps/LoRaMac/CMakeLists.txt:175 (generate_run_gdb_stlink):
  Unknown CMake command "generate_run_gdb_stlink".
```

Here is the following build commands that I used:
`XXX@laptop:/mnt/c/Users/XXXX/desktop/Electronics/LoRaMac-node-master/build$ cmake - DCMAKE_BUILD_TYPE=Release \
> -DTOOLCHAIN_PREFIX=""C:\toolchain"" \
> -DCMAKE_TOOLCHAIN_FILE="../cmake/toolchain-arm-none-eabi.cmake" \
> -DAPPLICATION="LoRaMac" \
> -DSUB_PROJECT="periodic-uplink-lpp" \
> -DCLASSB_ENABLED="ON" \
> -DACTIVE_REGION="LORAMAC_REGION_US915" \
> -DREGION_EU868="OFF" \
> -DREGION_US915="ON" \
> -DREGION_CN779="OFF" \
> -DREGION_CN779="OFF" \
> -DREGION_AU915="OFF" \
> -DREGION_AS923="OFF" \
> -DREGION_CN470="OFF" \
> -DREGION_KR920="OFF" \
> -DREGION_IN865="OFF" \
> -DREGION_RU864="OFF" \
> -DBOARD="NucleoL073" \
> -DMBED_RADIO_SHIELD="SX1276MB1MAS" \
> -DSECURE_ELEMENT="SOFT_SE" \
> -DSECURE_ELEMENT_PRE_PROVISIONED="ON" \
> -DUSE_RADIO_DEBUG="ON" ..`

@djaeckle 
I tried setting it up in visual studio code, but I get the following error compiler problem, do you know what I should look at to resolve it?
`CMake Error at C:/Program Files/CMake/share/cmake-3.19/Modules/CMakeTestCCompiler.cmake:66 (message):The C compiler

  "C:/Program Files (x86)/GNU Arm Embedded Toolchain/9 2020-q2-update/bin/arm-none-eabi-gcc.exe"

is not able to compile a simple test program.

It fails with the following output:

  Change Dir: C:/Users/XXXXXX/Downloads/LoRaMac-node-master/build/CMakeFiles/CMakeTmp
  
  Run Build Command(s):C:/msys64/mingw64/bin/mingw32-make.exe cmTC_798f0/fast && C:/msys64/mingw64/bin/mingw32-make.exe  -f CMakeFiles\cmTC_798f0.dir\build.make CMakeFiles/cmTC_798f0.dir/build
  mingw32-make[1]: Entering directory 'C:/Users/XXXXXX/Downloads/LoRaMac-node-master/build/CMakeFiles/CMakeTmp'
  Building C object CMakeFiles/cmTC_798f0.dir/testCCompiler.c.obj
  C:\PROGRA~2\GNUARM~1\92020-~1\bin\AR19DD~1.EXE    -o CMakeFiles\cmTC_798f0.dir\testCCompiler.c.obj -c "C:\Users\XXXXXX\Downloads\LoRaMac-node-master\build\CMakeFiles\CMakeTmp\testCCompiler.c"
  Linking C executable cmTC_798f0.exe
  "C:\Program Files\CMake\bin\cmake.exe" -E cmake_link_script CMakeFiles\cmTC_798f0.dir\link.txt --verbose=1
  "C:\Program Files\CMake\bin\cmake.exe" -E rm -f CMakeFiles\cmTC_798f0.dir/objects.a
  C:\PROGRA~2\GNUARM~1\92020-~1\bin\ARM-NO~2.EXE cr CMakeFiles\cmTC_798f0.dir/objects.a @CMakeFiles\cmTC_798f0.dir\objects1.rsp
  C:\PROGRA~2\GNUARM~1\92020-~1\bin\AR19DD~1.EXE -Wl,--whole-archive CMakeFiles\cmTC_798f0.dir/objects.a -Wl,--no-whole-archive -o cmTC_798f0.exe -Wl,--out-implib,libcmTC_798f0.dll.a -Wl,--major-image-version,0,--minor-image-version,0 
  c:/progra~2/gnuarm~1/92020-~1/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: unrecognized option '--major-image-version'
  c:/progra~2/gnuarm~1/92020-~1/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: use the --help option for usage information
  collect2.exe: error: ld returned 1 exit status
  mingw32-make[1]: *** [CMakeFiles\cmTC_798f0.dir\build.make:106: cmTC_798f0.exe] Error 1
  mingw32-make[1]: Leaving directory 'C:/Users/XXXXXX/Downloads/LoRaMac-node-master/build/CMakeFiles/CMakeTmp'
  mingw32-make: *** [Makefile:139: cmTC_798f0/fast] Error 2
  
  

CMake will not be able to correctly generate this project.`

Thanks,

closed 2020-12-09 by mluis1