Lora-net/LoRaMac-node #992: One byte transmissions over i2c

elsalahy, 2020-11-25:
**Summary:** 

Currently LMN provides two abstraction functions for i2c transmission.

https://github.com/Lora-net/LoRaMac-node/blob/2fbd31785aad6405693cf2f589ef7389bffeb687/src/system/i2c.h#L83

Which is designed to allow for one byte transmissions.

and 

https://github.com/Lora-net/LoRaMac-node/blob/2fbd31785aad6405693cf2f589ef7389bffeb687/src/system/i2c.h#L94

Which is designed to allow for multi byte transmissions.

The issue is that both interfaces force a minimum of 2 byte transmissions (addr (data address) + 1 byte in buffer)

**For example:**
In order to achieve this sequence

• Start Condition
• Device Address Byte
• Command/Word Address Byte (0x02 for example)
• Stop Condition

We would struggle with this function
```c 
I2cWrite( *obj, deviceAddr, addr (0x0), data (0x02));
``` 
or 
```c
I2cWrite( obj, deviceAddr, addr (0x02), data (NULL));
```

And as it can be observed neither is an ideal solution, especially the second one where we send a NULL buffer as most underlaying HAL implementation will report an error.

If we observe ST abstraction HAL as reference, we can observe two functions

```c
HAL_StatusTypeDef HAL_I2C_Mem_Write(I2C_HandleTypeDef *hi2c, uint16_t DevAddress, uint16_t MemAddress, uint16_t MemAddSize, uint8_t *pData, uint16_t Size, uint32_t Timeout)
```
and 

```c
HAL_StatusTypeDef HAL_I2C_Master_Transmit(I2C_HandleTypeDef *hi2c, uint16_t DevAddress, uint8_t *pData, uint16_t Size,
                                          uint32_t Timeout)
```
The second function allows for one byte transmissions as it doesn't force a MemAddress.

This is a trivial issue and we can solve it easily by creating multi byte transmission interface that doesn't take data address as an argument.

This issue became apparent in #981 solution where a contributor is trying to to handle one byte transmissions without any workarounds in the underlying layer.

@mluis1 let me know what you think, and I'm happy to help with implementing the new interface if we need one.

mluis1, 2020-11-25:
Yes, I think it is wise to have 2 separate API one to address directly the I2C bus and another to address memory based I2C devices.

Would the attached patch fulfill the requirements?
[i2c-api-update.patch.txt](https://github.com/Lora-net/LoRaMac-node/files/5596696/i2c-api-update.patch.txt)

elsalahy, 2020-11-26:
Looks great, thanks @mluis1 as this fixes the fundamental issue in #981

mluis1, 2020-11-26:
Thanks for the feedback @elsalahy.

I'll apply the proposed patch as soon as possible.

closed 2020-12-08 by mluis1