Lora-net/LoRaMac-node #993: LORAWAN 1.6 certification (5 seconds uplink frame)

lcharpenwi6labs, 2020-11-25:
Hello
In last LORAWAN certification 1.6, the device switched in test mode is supposed to send a counter message every 5 seconds.
I just run the example classA/B-L072Z-LRWAN1 and I see the device is sending every 20s when compliance is set to true.
What kind of mechanism in the stack is preventing the device to send every 5 seconds?
Thanks a lot for your help.

mluis1, 2020-11-25:
When using the provided example without modifications it should behave as expected while running the certification tests.

When running the certification tests the test tool will send `01010101` on port 224. Once the device receives this command it will enter in certification operation mode by setting `ComplianceTest.Running` variable to `true`.

Once in certification mode the `TxDutyCycleTime` is forced to `5000` ms => `5` seconds.

```c
            case DEVICE_STATE_CYCLE:
            {
                DeviceState = DEVICE_STATE_SLEEP;
                if( ComplianceTest.Running == true )
                {
                    // Schedule next packet transmission
                    TxDutyCycleTime = 5000; // 5000 ms
                }
                else
                {
                    // Schedule next packet transmission
                    TxDutyCycleTime = APP_TX_DUTYCYCLE + randr( -APP_TX_DUTYCYCLE_RND, APP_TX_DUTYCYCLE_RND );
                }

                // Schedule next packet transmission
                TimerSetValue( &TxNextPacketTimer, TxDutyCycleTime );
                TimerStart( &TxNextPacketTimer );
                break;
            }
```

This means that the `TxNextPacketTimer` will expire every 5 seconds and on its callback will set `DeviceState` to `DEVICE_STATE_SEND` and `NextTx ` to `true`.

```c
/*!
 * \brief Function executed on TxNextPacket Timeout event
 */
static void OnTxNextPacketTimerEvent( void* context )
{
    MibRequestConfirm_t mibReq;
    LoRaMacStatus_t status;

    TimerStop( &TxNextPacketTimer );

    mibReq.Type = MIB_NETWORK_ACTIVATION;
    status = LoRaMacMibGetRequestConfirm( &mibReq );

    if( status == LORAMAC_STATUS_OK )
    {
        if( mibReq.Param.NetworkActivation == ACTIVATION_TYPE_NONE )
        {
            // Network not joined yet. Try to join again
            JoinNetwork( );
        }
        else
        {
            DeviceState = DEVICE_STATE_SEND;
            NextTx = true;
        }
    }
}
```

Then when the main function loops again it will transmit the frame.

```c
            case DEVICE_STATE_SEND:
            {
                if( NextTx == true )
                {
                    PrepareTxFrame( AppPort );

                    NextTx = SendFrame( );
                }
                DeviceState = DEVICE_STATE_CYCLE;
                break;
            }
```

lcharpenwi6labs, 2020-11-25:
Thanks for your short reply.
Actually the problem came from the server that add a RX2 delay set to 5 seconds instead of 1 second...
It takes me some time to see this.

mluis1, 2020-11-25:
As it looks to be solved. Can we close the issue?

lcharpenwi6labs, 2020-11-25:
Yes you can. Thanks

closed 2020-11-25 by mluis1