Lora-net/LoRaMac-node #996: Fix STM platforms IRQ handlers when using a C++ compiler

mluis1, 2020-11-26:
Followup issue for pull-requests https://github.com/Lora-net/LoRaMac-node/pull/974 and https://github.com/Lora-net/LoRaMac-node/pull/975

The attached patch should fix the observed problem.
[cpp-irq-extern-c-wrapper.patch.txt](https://github.com/Lora-net/LoRaMac-node/files/5602244/cpp-irq-extern-c-wrapper.patch.txt)

@Xeenych Could you please verify this proposition?

Xeenych, 2020-12-02:
Sorry, I don't know how to read or use this patch.

mluis1, 2020-12-02:
On a terminal window go to the repository directory. Copy the patch file to that directory.
Then execute below commands in order to apply the patch. 1st we set the repository to master branch, 2nd we update the repository with latest changes on GitHub and then 3rd we apply the patch.

```bash
$ cd <path to your repository directory>/loramac-node
$ git checkout master
$ git pull -v
$ git apply cpp-irq-extern-c-wrapper.patch.txt
```

Once done you can compile your project. 
If using an IDE don't forget to move `sysIrqHandler.c` file to the right place in your project.

mluis1, 2020-12-02:
In case it is easier I have created pull request #998.

Could you please try it?

Xeenych, 2020-12-02:
Thanks. This works.
The new header sysIrqHandlers.h solves the issue.

mluis1, 2020-12-02:
Nice, thanks for the feedback.

PR #998 (2020-12-07): Fixed STM platforms IRQ handlers when using a C++ compiler
Closes #996
closed 2020-12-07 by mluis1