Lora-net/LoRaMac-node #999: SX126x: Use optimal PA settings according to datasheet 13.1.14

mluis1, 2020-12-02:
Please refer to closed pull-request #985 for further details.

mluis1, 2022-01-12:
@xanthio We are currently updating all radio drivers to newer versions. The SX126x PA settings management has been changed and checked by Semtech R&D.

As an example for NucleoL073 platform the SX126x PA management is done as follows
https://github.com/Lora-net/LoRaMac-node/blob/c12af2a9357c536edd49793181707b359b310be7/src/boards/NucleoL073/ral_sx126x_bsp.c#L87-L160

You can take a look at [v5.0.0-branch](https://github.com/Lora-net/LoRaMac-node/tree/v5.0.0-branch) in order to preview the changes that have been done. So far we successfully have run Class A pre-certification tests using every supported radio (SX126x, SX127x and LR1110).

For further information about the changes that will be introduced for v5.0.0 release please refer to #1239 discussion topic.

closed 2022-01-12 by mluis1

raj-muniraju, 2025-06-16:
> [@xanthio](https://github.com/xanthio) We are currently updating all radio drivers to newer versions. The SX126x PA settings management has been changed and checked by Semtech R&D.
> 
> As an example for NucleoL073 platform the SX126x PA management is done as follows
> 
> [LoRaMac-node/src/boards/NucleoL073/ral_sx126x_bsp.c](https://github.com/Lora-net/LoRaMac-node/blob/c12af2a9357c536edd49793181707b359b310be7/src/boards/NucleoL073/ral_sx126x_bsp.c#L87-L160)
> 
> Lines 87 to 160 in [c12af2a](/Lora-net/LoRaMac-node/commit/c12af2a9357c536edd49793181707b359b310be7)
> 
>  void ral_sx126x_bsp_get_tx_cfg( const void* context, const ral_sx126x_bsp_tx_cfg_input_params_t* input_params, 
>                                  ral_sx126x_bsp_tx_cfg_output_params_t* output_params ) 
>  { 
>      int8_t pwr = input_params->system_output_pwr_in_dbm; 
>   
>      output_params->pa_cfg.pa_lut = 0x01; 
>   
>  #if defined( SX1261MBXBAS ) 
>      // Clamp power if needed 
>      if( pwr > 15 ) 
>      { 
>          pwr = 15; 
>      } 
>      else if( pwr < -17 ) 
>      { 
>          pwr = -17; 
>      } 
>   
>      // config pa according to power 
>      if( pwr == 15 ) 
>      { 
>          output_params->pa_cfg.device_sel                 = 0x01;  // select SX1261 device 
>          output_params->pa_cfg.hp_max                     = 0x00;  // not used on sx1261 
>          output_params->pa_cfg.pa_duty_cycle              = 0x06; 
>          output_params->chip_output_pwr_in_dbm_configured = 14; 
>          output_params->chip_output_pwr_in_dbm_expected   = 15; 
>      } 
>      else if( pwr == 14 ) 
>      { 
>          output_params->pa_cfg.device_sel                 = 0x01;  // select SX1261 device 
>          output_params->pa_cfg.hp_max                     = 0x00;  // not used on sx1261 
>          output_params->pa_cfg.pa_duty_cycle              = 0x04; 
>          output_params->chip_output_pwr_in_dbm_configured = 14; 
>          output_params->chip_output_pwr_in_dbm_expected   = 14; 
>      } 
>      else 
>      { 
>          output_params->pa_cfg.device_sel                 = 0x01;  // select SX1261 device 
>          output_params->pa_cfg.hp_max                     = 0x00;  // not used on sx1261 
>          output_params->pa_cfg.pa_duty_cycle              = 0x04; 
>          output_params->chip_output_pwr_in_dbm_configured = pwr; 
>          output_params->chip_output_pwr_in_dbm_expected   = pwr; 
>      } 
>  #elif defined( SX1262MBXCAS ) | defined( SX1262MBXDAS ) 
>      // Clamp power if needed 
>      if( pwr > 22 ) 
>      { 
>          pwr = 22; 
>      } 
>      else if( pwr < -9 ) 
>      { 
>          pwr = -9; 
>      } 
>      if( pwr == 22 ) 
>      { 
>          output_params->pa_cfg.device_sel                 = 0x00;  // select SX1262 device 
>          output_params->pa_cfg.hp_max                     = 0x07;  // to achieve 22dBm 
>          output_params->pa_cfg.pa_duty_cycle              = 0x04; 
>          output_params->chip_output_pwr_in_dbm_configured = 22; 
>          output_params->chip_output_pwr_in_dbm_expected   = 22; 
>      } 
>      else 
>      { 
>          output_params->pa_cfg.device_sel                 = 0x00;  // select SX1262 device 
>          output_params->pa_cfg.hp_max                     = 0x07;  // to achieve 22dBm 
>          output_params->pa_cfg.pa_duty_cycle              = 0x04; 
>          output_params->chip_output_pwr_in_dbm_configured = pwr; 
>          output_params->chip_output_pwr_in_dbm_expected   = pwr; 
>      } 
>  #else 
>  #error "Please define a radio shield to be used" 
>  #endif 
>      output_params->pa_ramp_time = SX126X_RAMP_40_US; 
>  } 
> You can take a look at [v5.0.0-branch](https://github.com/Lora-net/LoRaMac-node/tree/v5.0.0-branch) in order to preview the changes that have been done. So far we successfully have run Class A pre-certification tests using every supported radio (SX126x, SX127x and LR1110).
> 
> For further information about the changes that will be introduced for v5.0.0 release please refer to [#1239](https://github.com/Lora-net/LoRaMac-node/discussions/1239) discussion topic.

@mluis1 
The previewed changes don't seem to match the table 13-21 for SX1262. We are evaluating our hardware, the data sheet has only few discrete value and not clear, Can you please provide more details on what PA setting to be used for each dB power setting?