Lora-net/LoRaMac-node #1013: SX126x: Perform image calibration after POR / cold sleep

rdaforno, 2020-12-16:
Image calibration must be performed after a POR or cold sleep as described in the datasheet on page 56.

In the [implementation](https://github.com/Lora-net/LoRaMac-node/blob/master/src/radio/sx126x/sx126x.c#L489) however, the image calibration is only performed once (variable `ImageCalibrated` is never reset to `false`).

I suggest to reset the variable `ImageCalibrated` in the following places:

-  `SX126xInit()`
-  `SX126xSetSleep()` (if `sleepConfig.Fields.WarmStart == 0`)

In principle it should also be reset whenever `SX126xReset()` is called.

mluis1, 2021-03-15:
Could you please review the proposition ?

rdaforno, 2021-03-15:
> Could you please review the proposition ?

looks good to me

closed 2021-03-15 by mluis1